MarlinFirmware/Configurations #201: Arduino does not build marlin 2.0.6.1 firmware

Subia75, 2020-09-04:
Hi everyone
i am trying to compile the firmware for my anycubic I3 mega S with marlin 2.0.6.1 before i had version 2.0.5 but i had not compiled it myself.

When I go to modify the configuration.h file to use my TFT35 display, arduino ide is no longer able to build the firmware and gives me an error.
The U8glib library are installed so I don't understand what that could be.

This is the Arduino repport:
"
  1.9.0-beta (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

fork / exec C: \ Users \ alessio.tius \ AppData \ Local \ Arduino15 \ packages \ arduino \ tools \ avr-gcc \ 7.3.0-atmel3.6.1-arduino7 / bin / avr-gcc.exe: Name of the file or extension too long.
Error compiling for board Arduino Mega or Mega 2560.
"

I am attaching a screenshot of the changes made to the configuration.h file compared to the pre-compiled one found on github.com.
[image]

I ask if anyone can help me.

Tkdmaster, 2020-09-04:
I also get this same error when building for my Tevo Tornado. I am also using the Android beta 1.9.0

Subia75, 2020-09-07:
Sorry for the delay.
I also tried the latest arduino 1.8.13 which is newer but doesn't build equally.

Honestly, it seems to be a problem when I select a different display, but I don't find any errors in disabling code strings, also because I'm comparing with 2.0.5 which builds me without problems.