MarlinFirmware/Configurations #203: [FR] Add support for Anet E12

andramadon, 2020-09-04:
There is support for the E10 and the E16 but not for the E12. Im not sure if i can just use either of those. I dont mean to noob it up. all of this code can be a bit intimidating.

thinkyhead, 2020-10-06:
Those might be a good starting point, but get tips from other users. And if you end up with a good config please submit it here!

Vertabreak, 2021-02-19:
if someone with the printer where to contact me and be will to do some testing im sure i could get this ported over to marlin 2.0.x.
if some existing older version can be provided that would speed along the process.
i can be reached on the marlin discord username vert with a blue name/role should someone wish to explore getting this done.
upon completion like most of my work i will contribute it back to upstream marlin making it available to everyone.
porting over printers from older versions of marlin is kinda my thing Ive ported over more then 20 printers.

Surfer2go, 2021-02-27:
Hi I have  a Anet E12 Printer runing with Marlin. If you need something Tested....

thinkyhead, 2022-04-10:
Duplicate of #649

closed 2022-04-10 by thinkyhead