MarlinFirmware/Configurations #204: Ender 3 Pro with 4.2.7 board

gxwilson69, 2020-09-06:
There are a lot of discussions [Ender 3 v1 with 4.2.7 board](https://github.com/MarlinFirmware/Marlin/issues/19029) and [Add Creality V4.27 Controller](https://github.com/MarlinFirmware/Marlin/pull/19088) but still no Marlin committed configs which take advantage for the Marlin/src/pins/stm32F1/pins_CREALITY_V427.h file. 

My attempts to get this to work on my own have been failures up to this point compiling and loading versions which have worked only partially. 

Each time I pull the examples/Creality/Ender-3 Pro V1.5 and then change the to be MOTHERBOARD BOARD_CREALITY_V427

My X endstop doesn't work and I still get the "Homing Failed" error message. 

I'm attempting to take a stock Ender 3 Pro and just upgrade to Marlin 2 using a Creality 4.2.7 board. No BLTouch or extras at this time.

seanwhiteau, 2020-09-26:
@gxwilson69 - I have an Ender 5 Pro, and a 4.2.2, and I had a "homing failed error" when I tried to compile Marlin 2.x bugfix for it.
In my case, I had to change:

**#define INVERT_X_DIR true
#define INVERT_Y_DIR true
#define INVERT_Z_DIR true**

to instead:
**#define INVERT_X_DIR false
#define INVERT_Y_DIR false
#define INVERT_Z_DIR false**

Also **#define INVERT_E0_DIR true**
worked for me as instead:
**#define INVERT_E0_DIR false**

Your problem might lie around the same configuration areas perhaps?

ellensp, 2020-09-26:
Take a look at current PR's

closed 2020-10-02 by gxwilson69