MarlinFirmware/Configurations #206: Creality CR-10 s4 configuration

JayJay2105, 2020-09-07:
Hi,

Is there a configuration file available for the CR-10 S4 which has a 400x400x400 print area,?

Thanks

JayJay2105, 2020-09-10:
In attempting to sort my own config files, I have discovered that with a fresh download and the 'standard' configuration files (both) everything compiles correctly,

 but if I use either the CR-10S or the CR-10-5S files I get the following error:

fork/exec C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc.exe: The filename or extension is too long.
Error compiling for board Arduino Mega or Mega 2560.

Using Arduino 1.8.13, I have searched the internet etc, and have added the build.path=c:\tmp to my preferences file and have moved the marlin folder to C:\ and even tried renaming it to just M

Any help greatly appreciated.

Arthurmaffre34, 2020-10-03:
you ave to use (visual studio code for example) with platformIO plugin it's a bug with arduino. I have the same printer

ellensp, 2020-10-05:
This is a Arduino IDE on Windows issue. Windows has a hard coded unchangeable max path/fork limit of 32K characters. Arduino IDE is written badly, when it attempts to link all the .o files together to form the final .bin file on large projects, such as Marlin, it generate one very long command that can easily exceed the windows limit.

There are a number of tricks to shorten the path under windows. 
Move marlin to root of drive, rename it to m
Delete unused parts of marlin such as unused HAL directories and unused display types. But this require in depth knowledge of how marlin works to know what is safe to remove.

Ultimately it is far better to use PlatformIO on vscode instead of Arduino IDE
PlatformIO only complies and links the code you need for your system, this avoids the Windows limitation. It also produces smaller binaries and is faster.

thinkyhead, 2020-10-06:
Please provide a configuration once you have adapted one to your liking from the S5 configuration, and we will publish it here.

thinkyhead, 2020-10-06:
Looks like we have a good config. Please submit any improvements that you find useful in your own setup.

closed 2020-10-06 by thinkyhead