MarlinFirmware/Configurations #207: Ender 3 v2 Endstops always reporting open - 2.0.6.1

kiwizznz, 2020-09-08:
Built the latest firmware (bin) using visual studio and the provided example default config for the Ender 3 v2 (Creality). 

Endstops always report open (also when pressing down):

Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: open

When homing, this causes the head/etc to smack into the end and keep going, messing up the belt. 

Have spent a few hours trying to work out why this is an issue (older firmware from Creality doesn't have this issue). I need to build my own as I have moved to a different extruder (needed to invert/reverse the extruder motor in firmware).

Configuration attached (its the only file I modified)

[Configuration.txt](https://github.com/MarlinFirmware/Configurations/files/5188894/Configuration.txt)

thinkyhead, 2020-09-10:
When I build 2.0.6.1 with your Configuration file it throws an error:
```cpp
 #error "LCD_BED_LEVELING requires a programmable LCD controller."
```

I have 2.0.6.1 installed on an Ender 3 V2 here (board v4.2.2) and all the endstops are working properly.

kiwizznz, 2020-09-10:
> When I build 2.0.6.1 with your Configuration file it throws an error:
> 
> ```c++
>  #error "LCD_BED_LEVELING requires a programmable LCD controller."
> ```
> 
> I have 2.0.6.1 installed on an Ender 3 V2 here (board v4.2.2) and all the endstops are working properly.

Thanks thinkyhead. Did you need to change anything from the default marlin settings for the Ender 3v2 when building your firmware? I'll turn off the LCD_BED_LEVELING as its not for this machine's screen. Someone else mentioned there were issues with endstops not working but that it was fixed in the bugfix release? I haven't had time to check again yet though.

thinkyhead, 2020-10-06:
The latest updates for the Ender 3 V2 are pretty good. Recommend you update to the latest Marlin code, or 2.0.7.1 when it comes out in a few days.

closed 2020-10-06 by thinkyhead