MarlinFirmware/Configurations #217: Allen key probe compile error "You can't home to a z min endstop with a Z_PROBE_ALLEN_KEY."

yankee14, 2020-09-18:
My [Configuration.h](https://gist.github.com/yankee14/e738ff03ac2fa38e093668fbf57fabb0) for reference.

I am playing around with making an allen key probe for my Ender 3 V1, but hit an error I'm a little confused about when compiling. I have wired a limit switch in series with my Z limit switch. I confirmed that everything is fine electrically because I can successfully home the printer with the switch in series. The error I get is:

```
In file included from sketch/src/HAL/DUE/fastio/../../../inc/MarlinConfig.h:41:0,
                 from sketch/src/HAL/DUE/fastio/G2_PWM.cpp:43:
sketch/src/HAL/DUE/fastio/../../../inc/SanityCheck.h:1570:4: error: #error "You can't home to a z min endstop with a Z_PROBE_ALLEN_KEY."
   #error "You can't home to a z min endstop with a Z_PROBE_ALLEN_KEY."
    ^~~~~
```
I have looked at the suggested [Delta configuration](https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/delta/generic/Configuration.h#L1043) as suggested in the configuration source file, and the way it is [written in the Marlin source](https://github.com/MarlinFirmware/Marlin/blob/9ade72aee94981b26be0ca125ab487166d0bd7e3/Marlin/src/module/probe.cpp#L159). I'm not sure if this is appropriate, since I plan on manually deploying and stowing the allen key with the probe pause functions elsewhere in the firmware.

Would someone be able to give suggestions about where I can look next?

[image]

closed 2020-09-29 by yankee14

thinkyhead, 2020-10-04:
Did you find the answer you were looking for?

yankee14, 2020-10-04:
@thinkyhead Unfortunately no, not yet.

thinkyhead, 2020-10-06:
If your machine is not a DELTA then simply comment out the `#error` line and carry on. The allen key probe is not supported for use on a non-delta machine, but you can probably ignore that.

yankee14, 2020-10-06:
@thinkyhead ah okay. I'll give it a shot, thanks