MarlinFirmware/Configurations #221: Ender 3 V2 problems

nossnahojse, 2020-09-21:
Done the BLT upgrade on my Ender 3 V2

Problem # 1 
I can't adjust the Z-offset while printing ???
 Why ?? Worked on stock Creality.
I can adjust it befor starting a print.

Problem # 2 
Prints starts at speed 447%. 
And even if i change it to 100% while heating up, it changes to 447% when starting print.
If i change when print started it keeps that speed during print.

Stalkeros000, 2020-09-23:
I was have that problem too after update FW from video tutorial from channel TeachingTech. Then I install latest FW from Smith3D and all work fine after that and that FW have much better menu options. Hope it help.

pixelicous, 2020-10-06:
Marlin 2.0.7,upgraded on top of latest ender 3 v2 official firmware. I also have issues. I used default ender 3 v2 config, when I stop a print it says "parking".. It homes to 0,0,0 then goes back to where I stopped it. If I move the Z up, it moves Z but also extrude the same amount of filament from the extruder, really quickly..

Restarting a failed print caused it to resume from same point even after renaming the file? Had to restart the machine

No bltouch or anything..

yysh12, 2020-10-20:
Why not just enable babystepping? That would allow for runtime changes to Z-offset.

pixelicous, 2020-10-20:
> I was have that problem too after update FW from video tutorial from channel TeachingTech. Then I install latest FW from Smith3D and all work fine after that and that FW have much better menu options. Hope it help.

smith3d is only if you have bltouch from what i saw.. i didnt have bltouch at the time.

the z parking issue is an actual issue as far as i understood from an issue i opened up