MarlinFirmware/Configurations #233: Hotend temperature show negative target values

wernju1, 2020-09-30:
Anycubic I3 mega / trigorilla board
Hotend temp. Shows negative target values.
e.g. preheat temp should 195° / display -67°

mytja, 2020-09-30:
Check the wiring!

wernju1, 2020-09-30:
With 2.0.6.1 and previous releases it works fine

wernju1, 2020-09-30:
The temperature while heating the hotend seems not to be affected and is displayed correctly.
What i mean is the target temperature, e.g. 180° for pla or 230° for abs.
These values are not displayed correct, therefore negative values were shown at the display instead.

mytja, 2020-10-01:
Hmmmmm. I'm out of ideas, maybe @thinkyhead knows the answer...

wernju1, 2020-10-01:
Problem is fixed with PR #19572

mytja, 2020-10-02:
Then okay...

ellensp, 2020-10-02:
@wernju1 Then perhaps you could close this issue?

wernju1, 2020-10-02:
Sorry, I

closed 2020-10-02 by wernju1