MarlinFirmware/Configurations #244: Rename config examples directory/folder "CR-10_5S" to "CR-10 S5" instead

Gamester17, 2020-10-05:
Please rename directory/folder "CR-10_5S" to "CR-10 S5" as Creality's 500x500x500mm CR-10 printer is named "S5" and, (not "5S").

https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality

Probably a typo but it regardless the directory/folder under config/examples is currently named wrongly:

https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/CR-10_5S

I would also like you to note that inconsistency of underscores used in some printer names there while others have spaces:

https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality

Extra inconsistent is the "CR-10mini" directory/folder which now doesn't have underscore or space in its name, it should have one.

Suggest you rename all Creality config examples directories/folders so all either has underscores or all have space in their names.

Reference:

https://www.creality.com/goods-detail/creality-cr-10-s5-3d-printer

https://www.creality3dofficial.com/products/creality-cr-10-s5-3d-printer

https://www.creality3d.shop/collections/3d-printer/products/creality-cr-10s-s5-3d-printer-diy-kit-large-printing-size-500x500x500mm

thisiskeithb, 2020-10-05:
We're accepting pull requests, so feel free to submit one to fix these issues 🙂

Gamester17, 2020-10-05:
OK, done so now ;)

closed 2020-10-06 by thinkyhead