MarlinFirmware/Configurations #247: Ender 5 Pro v4.2.2

somejavadev, 2020-10-05:
Hi, I bought a Creality Ender 5 Pro a month ago and it has shipped with a 32bit board v4.2.2 board. Creality does provide marlin 2.0.1 downloads for this this board: https://www.creality.com/download, would it be possible to provide a configuration for this board as well? I am not sure how compatible the v4.2.7 configurations would be for the v4.2.2 board.

ellensp, 2020-10-06:
I suspect all that is needed this change this to a V4.2.2 would be to change
#define MOTHERBOARD BOARD_CREALITY_V427
to
#define MOTHERBOARD BOARD_CREALITY_V4
and perhaps some axis inversions
Perhaps you could do some testing and this give this a try?

somejavadev, 2020-10-06:
@ellensp Thanks, let me play around with it. I am fairly new to Marlin so I must still get used to how everything works but the code seems really well documented.

somejavadev, 2020-10-06:
@ellensp, I gave it a try but for some reason I cannot get my BLTouch to work when compiling the firmware myself. Using the downloaded compiled version from Creality the BLtouch works fine. The xyz axis seem fine with the BOARD_CREALITY_V4 specified but I think there is something with the PINS and the BLTouch not behaving.

somejavadev, 2020-10-06:
@ellensp, thank you for the help. I took the base configuration for the Ender 5 Pro V4.2.7 and only changed the motherboard as you suggested to BOARD_CREALITY_V4. I am printing the XYZ calibration cube now and everything seems in order. This video helped https://www.youtube.com/watch?v=neS7lB7fCww&feature=youtu.be, as I originally used the BLTouch "kit" method of installing the BLTouch which is now redundant since the newer 4.2.2 board comes with additional ports.

closed 2020-10-06 by somejavadev