MarlinFirmware/Configurations #250: Opt-in MarlinFirmware Configurations repo to participate in Hacktoberfest?

Gamester17, 2020-10-05:
Any plans to allow this repo to participate in Hacktoberfest? If so then you need to opt-in by classifying this repository with the hacktoberfest topic, see

https://hacktoberfest.digitalocean.com/hacktoberfest-update

ellensp, 2020-10-06:
Please stop spamming this everywhere.

thinkyhead, 2020-10-06:
I participated two years ago and all I got was this very nice t-shirt.

closed 2020-10-06 by thinkyhead

Gamester17, 2020-10-06:
> Please stop spamming this everywhere.

I did not spam it "everywhere". I only posted the question to the two Marlin repositories that recently contributed pull requests to.

So this is perfectly valid questions here and not spam.

thinkyhead, 2020-10-06:
In fact, it is SPAM because we accept only Bug Reports and Feature Requests. For ANYTHING ELSE we provide the following guidance:

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Gamester17, 2020-10-06:
> In fact, it is SPAM because we accept only Bug Reports and Feature Requests. For ANYTHING ELSE we provide the following guidance:
> 
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. 

Let us agree to disagree then because I would argue that it is a "development-related issue" and not a "user-support question".

No worries, won't bother you again.

thinkyhead, 2020-10-06:
Thanks! We like to keep the traffic here limited to issues and features that we are working on.