MarlinFirmware/Configurations #262: Can't detect endstops

Doyle4, 2020-10-10:
Installed 2.0.7.1 onto my Artillery Genius.
I have had to invert my Y, but for some reason, when it homes it was going the wrong direction.
I changed Home Direction from -1 to 1, Y endstop from Min to Max.
Y_MIN_ENDSTOP_INVERTING changed to false from true, Y_MAX_ENDSTOP_INVERTING changed to True from false.

But my bed still crashes and I have to Kill the printer.

I reset eeprom after each flash.

Doyle4, 2020-10-10:
Also tried Y_MAX, still no detection on sensor, swapped sensor cable from Y- to Y+ port, still no luck.
I have had to go back to Y been inverted as nothing solved the issue

thinkyhead, 2020-10-14:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-14 by thinkyhead

Doyle4, 2020-10-14:
Sorry, I thought this was a bug as it was working fine before 2.0.7

Tried all methods provided, Y is still currently reversed.