MarlinFirmware/Configurations #265: [Discussion] how to handle the example config of an Board upgrade

Marlor, 2020-10-12:
I just saw the new config for the cr-10s with SKR Mini E3 2.0 upgrade and the problems with it https://github.com/MarlinFirmware/Configurations/pull/238#issuecomment-706703750

How should the Config for an Upgrade be handled?

Just the minimum? So the new Board is behaving as close as possible to the old one.
Essentially just changing `MOTHERBOARD`, `DRIVER_TYPE` and sometimes` INVERT_DIR` etc

What about "must have" Features like S`TEALTHCHOP/HYBRID_THRESHOLD` on a TMC Board

Or "nice to have" Features like `S_CURVE_ACCELERATION` and bigger Buffer on a stronger 32Bit Board

chepo92, 2020-10-12:
I think [Discussion] doesn't belong here, discord is more active
Regarding your question, I think the configs are just a basic/standard working default, any advanced feature should be enabled/customized by the user

Gamester17, 2020-10-19:
@Marlor Good feedback IMO, however, a better way forward would maybe be if you yourself make those edits and submit the changes as a separate/individual pull requests for review and PR discussion (somewhat similar to the current handling of https://github.com/MarlinFirmware/Configurations/pull/269 PR).

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10S/BigTreeTech%20SKR%20Mini%20E3%202.0

Maybe also take an extra look at CrealityV1 stock board too?

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10S/CrealityV1

Same goes for CR-10 which is based on the same config https://github.com/MarlinFirmware/Configurations/pull/237

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10/BigTreeTech%20SKR%20Mini%20E3%202.0

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10/CrealityV1

thinkyhead, 2020-10-21:
We can talk about this more on Discord, but my feeling is that we should provide a configuration that gives at least as many features as the original shipping machine, but should also add features that should be there, and should tune features that are a little bit off. For example, many lead-screw Z axis machines can move Z very fast, but the max and homing speeds may be too low. We can fix things like this in these configurations and offer any other improvements that help to fulfill the potential of the stock hardware.

closed 2020-10-21 by thinkyhead

Marlor, 2020-10-23:
Just thought it would be cleaner discussing a baseline for the configs here, instead of a chat.
I will have a look at Discord