MarlinFirmware/Configurations #266: Suggestion: Ender 3 Pro V1.5 should use Allegro by Default

vladipus, 2020-10-13:
Seems like during the transition phase (from 8 bits to 32bits boards) some Ender 3 users got the 4.2.2 board with TMC drivers. But the plan is that the Ender 3 printers have 4.2.2 boards with allegro drivers and only the V2 with TMC.

yysh12, 2020-10-20:
Yeah. My 4.2.2 board has HR4988's in them, but the config I had installed had it set to TMC2208's. Surprisingly didn't really notice a difference between the driver settings though. They both more or less worked the same? But yeah the default for the pros should definitely use the 4988's as default.

thinkyhead, 2020-10-23:
I will set the default to A4988 and add a `#warning` that will appear during the build.

closed 2020-10-23 by thinkyhead