MarlinFirmware/Configurations #272: ULTIPANEL vs IS_ULTIPANEL

kylegordon, 2020-10-16:
There seems to be an issue with ULTIPANEL and/or IS_ULTIPANEL that affects all boards.

I believe Configuration_adv.h#L1054 changed yesterday by @thinkyhead  should be #ifdef ULTIPANEL

I'm new to looking at Marlin, and whilst I'd put in a PR for changing all files, it seems configs come from somewhere else before being committed to git

cstueckrath, 2020-10-17:
my fix for lines 1051 ff:

`#if EITHER(IS_ULTIPANEL, EXTENSIBLE_UI)
  #define MANUAL_FEEDRATE { 50*60, 50*60, 4*60, 2*60 } // (mm/min) Feedrates for manual moves along X, Y, Z, E from panel
  #define SHORT_MANUAL_Z_MOVE 0.025 // (mm) Smallest manual Z move (< 0.1mm)
  #if ENABLED(IS_ULTIPANEL)
    #define MANUAL_E_MOVES_RELATIVE // Display extruder move distance rather than "position"
    #define ULTIPANEL_FEEDMULTIPLY  // Encoder sets the feedrate multiplier on the Status Screen
  #endif
#endif`

thinkyhead, 2020-10-21:
When something is defined as empty (e.g., `#define IS_ULTIPANEL`) it will make `#if IS_ULTIPANEL` angry. So, this has been fixed, and the configs in their current state must be used with Marlin 2.0.8 (or the current bugfix).

closed 2020-10-21 by thinkyhead