MarlinFirmware/Configurations #274: Ender 3 v2 with BTT SKR Mini E3 V2

Stalkeros000, 2020-10-18:
Hi all. I am trying to connect the board to this printer, used different configs for Marlin (Bigtreetech and Marlin), but everywhere I run into a problem.
If I specify board "default_envs = STM32F103RC" in the file platformio.ini (I also tried boards STM32F103RC_btt; STM32F103RC_BTT_512K), and in the file Configuration.h specify:
 #ifndef MOTHERBOARD
define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V2_0, 

then an error occurs during compilation:
*** [.pio\build\STM32F103RC_btt\src\src\lcd\dwin\e3v2\dwin.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt\src\src\lcd\dwin\e3v2\rotary_encoder.cpp.o] Error 1

[image]
[https://ibb.co/ydSn2Hx](screen)
Please give advice. Thanks.
Log:

`
> Executing task: C:\Users\Stalker\.platformio\penv\Scripts\platformio.exe run <

Processing STM32F103RC_btt (platform: ststm32@~6.1; board: genericSTM32F103RC; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 (6.1.1) > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 2.10000.200103 (1.0.0)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.91
|-- <Wire> 1.0
|-- <FreeRTOS701>
|-- <Servo(STM32F1)> 1.1.2
Building in release mode
Compiling .pio\build\STM32F103RC_btt\FrameworkArduinoVariant\board.cpp.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduinoVariant\wirish\boards.cpp.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduinoVariant\wirish\boards_setup.cpp.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduinoVariant\wirish\start.S.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduinoVariant\wirish\start_c.c.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduinoVariant\wirish\syscalls.c.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\Print.cpp.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\WString.cpp.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\cxxabi-compat.cpp.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\ext_interrupts.cpp.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\hooks.c.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\itoa.c.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\libmaple\adc.c.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\libmaple\adc_f1.c.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\libmaple\bkp_f1.c.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\libmaple\dac.c.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\libmaple\dma.c.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\libmaple\dma_f1.c.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\libmaple\exc.S.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\libmaple\exti.c.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\libmaple\exti_f1.c.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\libmaple\flash.c.o
Compiling .pio\build\STM32F103RC_btt\FrameworkArduino\libmaple\fsmc_f1.c.o
[... 499 lines cut ...]
 #define READ(IO)                (PIN_MAP[IO].gpio_device->regs->IDR & (1U << PIN_MAP[IO].gpio_bit) ? HIGH : LOW)
                                          ^~
Marlin\src\lcd\dwin\e3v2\rotary_encoder.cpp:80:7: note: in expansion of macro 'BUTTON_PRESSED'
   if (BUTTON_PRESSED(EN2)) newbutton |= 0x02;
       ^~~~~~~~~~~~~~
Marlin\src\lcd\dwin\e3v2\rotary_encoder.h:44:34: error: 'BTN_ENC' was not declared in this scope
 #define BUTTON_PRESSED(BN) !READ(BTN_## BN)
                                  ^
Marlin\src\lcd\dwin\e3v2\../../../inc/../HAL/./STM32F1/fastio.h:32:42: note: in definition of macro 'READ'
 #define READ(IO)                (PIN_MAP[IO].gpio_device->regs->IDR & (1U << PIN_MAP[IO].gpio_bit) ? HIGH : LOW)
                                          ^~
Marlin\src\lcd\dwin\e3v2\rotary_encoder.cpp:81:7: note: in expansion of macro 'BUTTON_PRESSED'
   if (BUTTON_PRESSED(ENC)) {
       ^~~~~~~~~~~~~~
*** [.pio\build\STM32F103RC_btt\src\src\lcd\dwin\e3v2\dwin.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt\src\src\lcd\dwin\e3v2\rotary_encoder.cpp.o] Error 1
======================================================= [FAILED] Took 88.27 seconds =======================================================

Environment      Status    Duration
---------------  --------  ------------
STM32F103RC_btt  FAILED    00:01:28.271
================================================== 1 failed, 0 succeeded in 00:01:28.271 ==================================================
The terminal process "C:\Users\Stalker\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.`

Stalkeros000, 2020-10-18:
HIDE or SPOILER not working, sorry, its my first message.

thisiskeithb, 2020-10-19:
The Ender-3 V2's LCD is not supported on other boards at this time.

jernejp21, 2020-10-20:
> 
> 
> The Ender-3 V2's LCD is not supported on other boards at this time.

Yes it is. You have to modify LCD connection yourself and add a bit of code. See [#424](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/424).

See this issue where I have explained what code you have to add to `pins_BTT_SKR_MINI_E3_common.h` to define all the pins for display, like BEEPER_PIN. That's why you get compiler errors. You didn't define BEEPER_PIN.

thisiskeithb, 2020-10-21:
Most hardware can be compatible when modded or code is added. I just said it’s not supported on other boards at this time.

...until code is added

> The Ender-3 V2's LCD is not supported on other boards **at this time**.

thinkyhead, 2020-10-23:
We shall continue to work on the DWIN LCD for Ender 3 V2 as time allows. We expect to have the full menu system implemented on this display Real Soon Now.

closed 2020-10-23 by thinkyhead