MarlinFirmware/Configurations #305: Marlin compilation error for skr v1.4 turbo with Ender 3 v2 screen DWIN_CREALITY_LCD

Azhaguvel19, 2020-10-30:
I have replaced my ender 3 v2 board with skr V1.4 turbo and using the same ender 3 v2 display but I am not able to compile the marlin code for it. I am getting the following errors. It compiles fine if I comment #define DWIN_CREALITY_LCD and enable #define REPRAP_DISCOUNT_SMART_CONTROLLER. It compiles and the board is working with rerap discount controller which I had as a spare.

Marlin\src\lcd\dwin\rotary_encoder.cpp:52:9: error: 'BEEPER_PIN' was not declared in this scope
Marlin\src\lcd\dwin\rotary_encoder.h:46:34: error: 'BTN_EN1' was not declared in this scope
Marlin\src\lcd\dwin\rotary_encoder.h:46:34: error: 'BTN_EN2' was not declared in this scope
Marlin\src\lcd\dwin\rotary_encoder.h:46:34: error: 'BTN_ENC' was not declared in this scope
Marlin\src\lcd\dwin\dwin.cpp:2024:11: error: 'strcat_P' was not declared in this scope; did you mean 'strcat'?
Marlin\src\lcd\dwin\dwin.cpp:2140:5: error: 'buzzer' was not declared in this scope
Marlin\src\lcd\dwin\dwin.cpp:2145:5: error: 'buzzer' was not declared in this scope

[image]

Seems this is because skr v1.4 board pins file does not have any Configs for DWIN_CREALITY_LCD.

Azhaguvel19, 2020-10-30:
tried this then I get the str function error.

[image]

[image]

Azhaguvel19, 2020-10-30:
If I change that then I end up with this error

[image]

thisiskeithb, 2020-10-31:
The stock Creality Ender-3 V2 LCD is not supported on other boards at this time.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-31 by thisiskeithb