MarlinFirmware/Configurations #310: Setting the steps per mm for the X2 stepper when using Dual X Carriage feature.

hallphilip, 2020-11-01:
X1 steps are set and working fine, X2's step count is doubled and I can't find a way to adjust it. Please any help would be great

sjasonsmith, 2020-11-01:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-01 by sjasonsmith

hallphilip, 2020-11-02:
I have looked for info on this everywhere, and I never had issues with earlier releases. I think it is possibly a bug is why I was asking here. Thanks anyhow.

sjasonsmith, 2020-11-02:
Even if there is a bug here, the entire issue template was ignored. Without the information it requires, there is no way to even try to answer your question.

You very likely have your two steppers configured for different numbers of micro steps. It is not possible for Marlin to use different numbers of steps/mm on one axis.