MarlinFirmware/Configurations #311: Request: Ender 6

pandaero, 2020-11-02:
Looking for examples of a config for the new Ender 6, any help?

gazalos, 2020-11-09:
Follow, I need it to

cwtechnh, 2020-12-01:
I also have a new Ender 6 with improper firmware.  Willing to help test firmware.  I also run Octoprint

thisiskeithb, 2020-12-01:
You'll have to ask Creality for the source. If you drop a .zip with the source/configs here someone can look at porting it over assuming they didn't do anything too wild with the main Marlin codebase.

ellensp, 2020-12-01:
first thing would to be take images of the motherboard, see if it one we already know or something new

cwtechnh, 2020-12-02:
> first thing would to be take images of the motherboard, see if it one we already know or something new

I will get a better pic of the whole board to post Friday,  but here's a quick peek

[image]

ellensp, 2020-12-02:
Yip they have done it again, another completely undocumented motherboard from creality
We know of V4.2.2, V4.2.7, V4.5.2  but this is a V4.3.1
This mean we cant proceed till someone maps out this controller. Which means getting access to one.

sjasonsmith, 2020-12-03:
Lookin go at the pictures they have clearly reworked the power supply portion of the board and added lots of passive components around the stepper and display connectors. Hopefully this is just a robustness upgrade and not a complete re-mapping of processor pins.

ellensp, 2020-12-03:
I notice the bltouch port... only the two IO pins are connected. I wonder to what....

cwtechnh, 2020-12-03:
I will get detailed picts tomorrow.  Thanks for all the responses.  I also emailed Creality to hopefully light a fire under them.

pandaero, 2020-12-04:
Hi, I got in touch with Creality, they sent me firmware files already in binary format, as well as a picture of the mainboard attached, let me know if this would help.
[image]

thisiskeithb, 2020-12-04:
They need to provide you the source code since Marlin is released under the GPL.

cwtechnh, 2020-12-05:
Here's actuals.  Did my best to get details through the tangle of wires that I didn't want to remove because everything was hot glued.

[image]
[image]
[image]
[image]
[image]

ellensp, 2020-12-05:
Good effort!
On the Probe plugs (white plug, 5 pins, 2 used) beside rainbow cable, What does the label on that wire say?
What sort of probe does this printer have?

pandaero, 2020-12-05:
> 
> 
> Good effort!
> On the Probe plugs (white plug, 5 pins, 2 used) beside rainbow cable, What does the label on that wire say?
> What sort of probe does this printer have?

One of them seems to say PWM, I'd assume the other is gnd? Though the PWM is connected in the pin labelled "out", and the other in the pin labelled "in"

ellensp, 2020-12-05:
That is why i'm quite curious to find out where those two cables go...  normally its a bltouch and IN is the PWM signal and out is the triggered signal.    So what are they using it fr this time?

pandaero, 2020-12-05:
well, one of the features of this printer is a filament break sensor, could it be that?

cwtechnh, 2020-12-05:
All of the board connections go to a breakout board mounted at the top corner of the XY gantry.  I have a bltouch attached and plugged into that board.  The firmware that Creality had ( then disappeared ) on their web site supported the bltouch.  But the firmware seems incomplete & octoprint throws an warning error when I connect up.  I have successfully done a auto level, which seemed to work and printed a calibration cube.

pandaero, 2020-12-05:
here's their firmware files, note the names for setups.
[Firmwares.zip](https://github.com/MarlinFirmware/Configurations/files/5646509/Firmwares.zip)

cwtechnh, 2020-12-05:
Thanks, will take a look at them in the am.  I’ll let you know those are the same ones I have.

fape, 2020-12-10:
Is the (micro)usb port useable as serial, and can we use to connect to eg octoprint? 
Because Creality support said "Endrer-6 has no USB interface and does not support online printing for the time being." and "This is a reserved port on the motherboard. The Ender-6 model has no USB function for the time being"

cwtechnh, 2020-12-10:
I am plugged into octoprint and used the terminal to set z offset for BLTouch when I found no way to do this in the current firmware. I also printed a calibration cube from octoprint.  Seems to work.

cwtechnh, 2020-12-10:
> here's their firmware files, note the names for setups.
> [Firmwares.zip](https://github.com/MarlinFirmware/Configurations/files/5646509/Firmwares.zip)

Those were the same firmwares that I had.

pandaero, 2020-12-18:
> 
> 
> They need to provide you the source code since Marlin is released under the GPL.

yeah, they don't seem to grasp that very well...

ellensp, 2020-12-18:
Blatantly disregarding the licensing you mean.
They have been told repeatedly, even by people who speak their language natively. 
So it not a language barrier issue. 
It straight out disrespect at this point.

ellensp, 2020-12-18:
I've just managed to get the provided Ender 6 firmware running on a dev board.
Its not very stable, but should provide some data
Ill be adding pins as I find them here 
https://docs.google.com/spreadsheets/d/1DYhh9fwLnvZzuNvMoBen9Dl68KN-4TWkbO94lJ4iwe4/edit#gid=0

ellensp, 2020-12-18:
M503 results
01:05:36.976 > echo:  G21    ; Units in mm (mm)
01:05:36.979 > 
01:05:36.979 > echo:Filament settings: Disabled
01:05:36.982 > echo:  M200 D1.75
01:05:36.982 > echo:  M200 D0
01:05:36.984 > echo:Steps per unit:
01:05:36.984 > echo: M92 X80.00 Y80.00 Z400.00 E140.00
01:05:36.990 > echo:Maximum feedrates (units/s):
01:05:36.993 > echo:  M203 X500.00 Y500.00 Z10.00 E100.00
01:05:36.996 > echo:Maximum Acceleration (units/s2):
01:05:36.999 > echo:  M201 X5000.00 Y5000.00 Z100.00 E10000.00
01:05:37.004 > echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
01:05:37.010 > echo:  M204 P2000.00 R3000.00 T3000.00
01:05:37.012 > echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
01:05:37.023 > echo:  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.40 E5.00
01:05:37.029 > echo:Home offset:
01:05:37.032 > echo:  M206 X0.00 Y0.00 Z0.00
01:05:37.035 > echo:Auto Bed Leveling:
01:05:37.037 > echo:  M420 S0 Z0.00
01:05:37.037 > echo:PID settings:
01:05:37.040 > echo:  M301 P30.86 I3.10 D75.84
01:05:37.043 > echo:  M304 P327.11 I19.20 D1393.45
01:05:37.046 > echo:Power-Loss Recovery:
01:05:37.048 > echo:  M413 S1
01:05:37.048 > echo:Z-Probe Offset (mm):
01:05:37.051 > echo:  M851 X0.00 Y10.00 Z0.20

M115 results
01:08:31.089 > FIRMWARE_NAME:Marlin Ver 1.0.2 (GitHub) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
01:08:31.106 > Cap:SERIAL_XON_XOFF:0
01:08:31.109 > Cap:BINARY_FILE_TRANSFER:0
01:08:31.109 > Cap:EEPROM:1
01:08:31.112 > Cap:VOLUMETRIC:1
01:08:31.112 > Cap:AUTOREPORT_TEMP:1
01:08:31.115 > Cap:PROGRESS:0
01:08:31.117 > Cap:PRINT_JOB:1
01:08:31.117 > Cap:AUTOLEVEL:1
01:08:31.120 > Cap:Z_PROBE:1
01:08:31.120 > Cap:LEVELING_DATA:1
01:08:31.123 > Cap:BUILD_PERCENT:0
01:08:31.123 > Cap:SOFTWARE_POWER:0
01:08:31.126 > Cap:TOGGLE_LIGHTS:0
01:08:31.128 > Cap:CASE_LIGHT_BRIGHTNESS:0
01:08:31.131 > Cap:EMERGENCY_PARSER:0
01:08:31.131 > Cap:PROMPT_SUPPORT:0
01:08:31.134 > Cap:AUTOREPORT_SD_STATUS:0
01:08:31.137 > Cap:THERMAL_PROTECTION:1
01:08:31.139 > Cap:MOTION_MODES:0
01:08:31.139 > Cap:CHAMBER_TEMPERATURE:0

ellensp, 2020-12-20:
From https://reprap.org/forum/read.php?415,879720 
The V4.3.1 is the same as a V4.2.2 except you have to swap X and Y step and direction pins to get the corexy moving in the correct directions

ellensp, 2020-12-20:
once this PR is merged to add this board https://github.com/MarlinFirmware/Marlin/pull/20512
Example config can be added

ellensp, 2020-12-29:
This has now been added to bugfix branch and will be in future releases

closed 2020-12-29 by ellensp