MarlinFirmware/Configurations #314: Config Files are "old"

mtcmapbell1, 2020-11-05:
I just spent forever, like the 5th time, going through and editing the config.h and .adv.h and now I'm getting an error say the "config file is too old; please update file before compiling. "

Can someone please advise how to fix this?  The config files say they're 2 days old.

thanks

thisiskeithb, 2020-11-05:
Start with a fresh download of Marlin 2.0.7.2 from https://marlinfw.org/meta/download/ and use a diff program like WinMerge or Meld to transfer your old settings to the new configs.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-05 by thisiskeithb

ellensp, 2020-11-05:
If you download config files for a version of marlin older than you have, it will give that error, regardless if you download it yesterday or not, wrong version is wrong version. 

It checks the CONFIGURATION_H_VERSION and CONFIGURATION_ADV_H_VERSION, but just changing that will end it tears, it indicates the config file has changed a lot from previous versions.

Compare the CONFIGURATION_H_VERSION of the config file with the original CONFIGURATION_H_VERSION in the marlin 
you downloaded.