MarlinFirmware/Configurations #315: Other Boards

DavidDaniels04, 2020-11-06:
Hello,

I was really hoping with how popular the Ender 5 Plus has become that there would both similiar options like within the Ender 3 Pro folder of other boards. Specifically btt skr 1.4 Turbo w/ 2209s and maybe even perhaps using 2130's too? lol..

ellensp, 2020-11-06:
Was looking into this. There are multiple issue to overcome.
Firstly the TFT touch screen. Firmware wise It is only compatible with the 8 bit AVR platform. It has a function SERIAL_GET_TX_BUFFER_FREE which only exists in AVR HAL 
Secondly It is normally plugged into Serial 2, pins 16 (TX) and 17 (RX) On EXP1 on a RAMPS.
There is no hardware serial port on EXP1 on the skr 1.4
There is a hardware serial port (1 on P0_15 and P0_16) on EXP2, but these are not the correct pins location. 
You would need to make a custom TFT cable  

P0_15 | P0_16
-- | --

P0_15 | P0_16
-- | --

P0_15 | P0_16
-- | --

DavidDaniels04, 2020-11-06:
Yep from my understanding a rewiring of the tft would need to be done in addition to board OR an upgrade to a btt tft style tft (which is what I am currently using).. I currently have a bondtech bmg and e3d v6 hotend direct drive setup. I'm more or less surprised to see the other printers have a variety of examples using other boards and only the

ellensp, 2020-11-06:
The pins are labeled on various images, like this eg. So this helps a little.   
[image]

ellensp, 2020-11-10:
Could you change the subject to something more specific perhaps "Ender 5 Plus, other board options".

earth4s, 2020-11-29:
I'm trying to get the SKR E3 Turbo working on my Ender 5 plus but I am using their TFT70 touchscreens instead of the stock screen. The main issue I have run into is the the BLTouch will not respond to commands. Not sure if its an issue with the nightly builds or something else.

Cjs10mm, 2020-12-11:
How hard would it be to get the info on how the TFT interfaces with the board from two users who have already made it work with non stock boards? Insanity Automation did it for Tiny Machines and DodgeDeBoulet did it for the SKR 1.4 Turbo and E3 Mini v2.

NezSez, 2021-03-16:
@earth4s For SKR 1.4 Turbo I had to switch wires on the BLTouch which came with my E5+...Creality swapped two of the wires in the group of 3 (not the black/white pair) to plug into their Creality V2.2 board that mine came with. Sry I don't remember which two for sure but I think it was NPWR and 2.0 on the Servo port. YMMV.

NezSez, 2021-03-19:
Given the hardware/software dependency permutations, other than having a web page/app, or local script (like `menuconfig` (et. al.) in Linux Kernel src) which could parse dependencies allowing customization with option dependency tracking, parsing #defines and displaying code comments/help/caveats, guiding the user in real time as it were, anything I can think of to optimize the current structure just ends up in "different only the same" category, or worse :)

This problem is why menuconfig was created, same for PKG leading to RPM, APT, YUM, and DNF.

I know that similar issues have been debated before many times and rejected in the past for obvious reasons [#21003 [RFC] Rename the bugfix branch](https://github.com/MarlinFirmware/Marlin/issues/21003).

Creating some example configs for the E5+ with SKR 1.4/T and various hardware will help many of course, but short of going the full-monty #define dep parsing I think it wouldn't benefit nearly as much, nor offer the ROI as the aforementioned page/app/script would. It seems like one could build off the existing sanitycheck scripts and the Marlin Autobuild to do so, and I imagine that both of those were designed with this goal in mind in addition to the primary functions they currently serve.

thisiskeithb, 2021-03-19:
> and I imagine that both of those were designed with this goal in mind

It'll get there eventually...

[image]

thisiskeithb, 2022-04-11:
Duplicate of https://github.com/MarlinFirmware/Configurations/issues/649

SteevieB, 2023-08-21:
I'm trying to get the BIGTREETECH SKR Mini E3 V3.0 working on my Ender 5. Can anyone help me point me in the right direction what do i have to adjust to get it working with Marlin 2.1.2.1 based on the E3 V2.0 config files?

thisiskeithb, 2023-08-21:
@SteevieB: This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

SteevieB, 2023-08-21:
@thisiskeithb Sorry, thank you very much for the helpful links!