MarlinFirmware/Configurations #317: Marlin does not compiles,Marlin 2.0x to a SKR1.4

DominG0S, 2020-11-06:
> Executing task: C:\Users\Usuario\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1768; framework: arduino)
------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
ModuleNotFoundError: No module named 'platformio.managers.package':
  File "C:\users\usuario\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168:
    env.SConscript(item, exports="env")
  File "C:\Users\Usuario\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Usuario\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\FIRmware de marlin\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 12:
    from platformio.managers.package import PackageManager
================================== [FAILED] Took 0.63 seconds ==================================

Environment    Status    Duration
-------------  --------  ------------
LPC1768        FAILED    00:00:00.632
============================ 1 failed, 0 succeeded in 00:00:00.632 ============================ 
The terminal process "C:\Users\Usuario\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

rhapsodyv, 2020-11-07:
You need to use a updated version of marlin. Use 2.0.7.2 or bugfix.

ellensp, 2020-11-07:
platformio.managers.package was removed in current  platformio. You need current common-dependencies.py https://raw.githubusercontent.com/MarlinFirmware/Marlin/2.0.x/buildroot/share/PlatformIO/scripts/common-dependencies.py

closed 2020-11-07 by ellensp