MarlinFirmware/Configurations #319: SKR 1.4 does not appears as a drive

DominG0S, 2020-11-07:
Installed marlin 2.0,but now I can't upload to the SKR anymore since it now doesn't appears as a drive

sjasonsmith, 2020-11-07:
Most likely you need to unmount the SD card from Marlin through the LCD or the M22 command. Host and Marlin cannot access storage at the same time.

This issue is being closed because it is incomplete. When reporting an issue, please be sure to follow all instructions at the top of the template, including debugging steps, and leave the sections headings intact.

For reference, these are the instructions included in the template:

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

closed 2020-11-07 by sjasonsmith

sjasonsmith, 2020-11-07:
It was just pointed out to me that this is on the Configurations repo, which does not have an issue template like the main firmware repo has.

In addition to unmounting, you will want to be check SDCARD_CONNECTION in Configuration_adv.h to be sure it matches the SD slot you want access to.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).