MarlinFirmware/Configurations #320: linear advance, TMC 2208 UART connexion error on SKR 1.3 . New homers tarantula RS

moriamoria, 2020-11-07:
Hello,

I try to use linear advance with my new printer, result extruder stop working ! very simple to reproduce, just on a TFT maker base extrude 3 times 5 mm.

So reading on the net, I try to uses UART. I follow 

https://www.instructables.com/UART-This-Serial-Control-of-Stepper-Motors-With-th/

tutorial, but I have TMC connexion error.

So I have done a test with Arduino and my TMC2208, and after soldering UART pin, it works!

[image]
[image]

[image]

![20201107 TMC_22088-UART_SOLDER](https://user-images.githubusercontent.com/25781747/98452471-f6b32280-
[TMC2208_uart_test.ino.txt](https://github.com/MarlinFirmware/Configurations/files/5505566/TMC2208_uart_test.ino.txt)

I have set skr 1.3 pin as follow 

https://3dprintbeginner.com/wp-content/uploads/2019/05/SKR-1.3-UART-guide.jpg

And I have also check with my scope on Rx, Tx, that I have good signal on UART TMC2208 PIN

So  if you could help me. Thanks

Regards to all

[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Configurations/files/5505572/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Configurations/files/5505573/Configuration.h.txt)
[pins_BTT_SKR_V1_3.h.txt](https://github.com/MarlinFirmware/Configurations/files/5505575/pins_BTT_SKR_V1_3.h.txt)

sjasonsmith, 2020-11-07:
Duplicate of #16076

sjasonsmith, 2020-11-07:
@moriamoria at this point this is just a known issue. It has been documented for over a year, it just depends on someone spending the time to try to rework linear advance to avoid this problem.

closed 2020-11-07 by sjasonsmith