MarlinFirmware/Configurations #333: GitHub Desktop - bugfix branch always shows commits need pushing and has unrelated history

charakaSamawry, 2020-11-18:
Every day, or time i load up, github desktop thinks it has 3 commits to push and 3 to pull and says there's an error because "Unable to merge unrelated histories in this repository".

Probably something im doing wrong but am at a loss, so just looking for help so i can more easily reference the bugfix config examples.

Thanks

ellensp, 2020-11-18:
These configs are basically blown away an recreated by scripts. Not actually updated. 
Best idea is to delete and re git clone them.

charakaSamawry, 2020-11-18:
I was just about to to add that currently i have to do what you suggested but thought, maybe, because im a complete novice, i was doing something wrong.

Thanks will close the issue then, didnt realise it was automated tbh, and i dont understand much of how github works, less and less everyday lol

closed 2020-11-18 by charakaSamawry

charakaSamawry, 2020-11-18:
@ellensp IS there any difference between the bugfix branch example configs and the 2.0.x branch ones? im guessing the answer is yes? hence i keep using the bugfix examples

sjasonsmith, 2020-11-18:
The ones with a complete version number don’t change.

The bugfix branch gets updated occasionally, and should usually work with the bugfix branch. As ellensp said these are not done as normal git branches and get reset each time they are updated.

The import branch is where changes are made, and will have brand new changes not yet in the bugfix branch.

charakaSamawry, 2020-11-19:
Thank you for confirming and explaining that