MarlinFirmware/Configurations #338: creality cr-20 wrong stepper directions?

ghylander, 2020-11-22:
in my printer the stepper directions were wrong in the config file, making the steppers go the other way around

the config that worked for me was

`// Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
#define INVERT_X_DIR false
#define INVERT_Y_DIR false
#define INVERT_Z_DIR true`

original was 
`// Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
#define INVERT_X_DIR true
#define INVERT_Y_DIR true
#define INVERT_Z_DIR false`

i did not at any point touch the motor connectors

sjasonsmith, 2020-12-01:
This is what is present in the CR-20 firmware you can download from Creality:
```
// Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
#define INVERT_X_DIR false
#define INVERT_Y_DIR false
#define INVERT_Z_DIR true

// @section extruder

// For direct drive extruder v9 set to true, for geared extruder set to false.
#define INVERT_E0_DIR false
#define INVERT_E1_DIR false
#define INVERT_E2_DIR false
#define INVERT_E3_DIR false
```

This also seems to match what I see in other CR-20 configurations.
Is there any possibility that you don't have an original motherboard?

I am hesitant to just switch these, since there seem to be multiple sources claiming the examples are correct.

ghylander, 2020-12-01:
unless it came with a motherboard other than the original, it's using the
original
for the record, once i made these modifications, built the firmware and
pushed it to the printer, when homing the Z axis was inverted too
i understand that one testimony is not enough, so i guess put it on hold?
i'll see if i can get more cr 20 non pro owners to upgrade the firmware
maybe it's just a build quality issue where some printers are just inverted
for whatever reason

thisiskeithb, 2020-12-01:
This may be another example of Crealtiy using different motherboard/drivers with no way easy way to tell which one they have installed.

ghylander, 2021-11-09:
closing this i suppose

closed 2021-11-09 by ghylander