MarlinFirmware/Configurations #354: [BUG] Example import-2.0.x/config/examples/delta/Tevo Little Monster wrong board defined

lemmingDev, 2020-12-02:
### Bug Description

The the example configuration for Tevo Little Monster, the MOTHERBOARD is defined as BOARD_MKS_GEN_L

The motherboard that comes with all Tevo Little Monsters is the MKS SBase, so the definition should be BOARD_MKS_SBASE

### My Configurations

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5279144/Configs.zip)

### Steps to Reproduce

1. Use the provided Tevo Little Monster Configuration.h and Configuration_adv.h files

**Expected behavior:** [What you expect to happen]
#define MOTHERBOARD BOARD_MKS_SBASE

**Actual behavior:** [What actually happens]
#define MOTHERBOARD BOARD_MKS_GEN_L

#### Additional Information
There could be other changes needed as well,

thinkyhead, 2021-04-15:
There are now V1 and V2 Tornado configs. Shouldn't be hard to add a V3 there and two more versions for the little monster too.

lemmingDev, 2022-01-09:
> @lemmingDev I attempted to update this for you, but I am not comfortable with the changes I have to make, without having the printer to test it.
> 
> Do you have a working configuration for a stock Little Monster?
> 
> The details that I am unsure about:
> 
> * Microsteps? 32 or 64? (And the steps/mm derived from this)
> * Axis and endstop inversion?
> * BLTouch configuration

Hi

I have successfully got the latest Marlin working on a stock TEVO Little Monster will all features working fine.

I had to edit a couple of pins from the MKS SBASE pins file as the thermistors were wrong for the way this was wired...

All required files included in the attached zip. Would be fantastic if this could be added to the official release

[Marlin Files.zip](https://github.com/MarlinFirmware/Configurations/files/7834932/Marlin.Files.zip)

thinkyhead, 2022-01-26:
Unfortunately the file download link is broken.

lemmingDev, 2022-01-26:
Are you sure?
I was just able to download the zip file above and extract the 3 files from inside...
In any case, here they are again
[Marlin.Files (1).zip](https://github.com/MarlinFirmware/Configurations/files/7945534/Marlin.Files.1.zip)

closed 2022-02-16 by thinkyhead

docman11, 2025-05-11:
@lemmingDev, I can't find stock TFT28 specified in your configuration.h - what type of screen are you used?