MarlinFirmware/Configurations #355: Ender 5 Pro now shipping with v4.2.2 mainboard

dougbalish1, 2020-12-02:
Dont believe this has been addressed, if it has, please link me to where:

The default ender 5 pro config now no longer completely works for (what i assume to be) most ender 5's shipping. I, and a few people ive been in contact with, have received ender 5's with creality v4.2.2 mainboards, and while creality has a .bin that works with both the stock version and bltouch (i have a bltouch) version on their site, they refuse to share the (marlin v2 based) source. 

I would think at least a warning in the config for ender 5 pros is in order to prevent wasted time, as well as possibly an ender 5 pro v1.5 type addition to the configs. 

I would love to help contribute, but i am new to github and open source (competent enough c programmer/debugger though) so if anyone wants to point me in the right direction to help, i would be glad. 

Some additional info in case it helps anyone: I have tried building the ender 5 pro config but using the STM32F103RET6//Creality board V4 settings transplanted into the ender 5 pro config, and i can get this to work with some tweaks, but not the bltouch. This indicates (to me at least) that the pinout is different. If anyone would like me to help debug also, i have access to a cheap oscope and multimeter, fpga, logic analyzer, etc.

ellensp, 2020-12-02:
Anyone know the pinout with option 3, never seen it.   Otherwise is fairly easy to setup 3 options and force thr user to chose one of them when bltouch is enabled

thisiskeithb, 2020-12-02:
> Anyone know the pinout with option 3, never seen it. Otherwise is fairly easy to setup 3 options and force the user to choose one of them when bltouch is enabled

I think they are referring to a "Pin 27 board" that was used on the 1.x boards to break out PIN 27 for the BLTouch or filament runout sensor (depending on config): https://www.amazon.com/Creality-BLTouch-Filament-Print3d-LLC/dp/B07L2Q2SSD

sjasonsmith, 2020-12-02:
It makes me sad that Creality put any customer in the position where optiobn #1 is not the correct answer.
It is clearly the "correct" way to do it for these boards. I wonder how many people it impacts if we just left that as the one and only example.

ellensp, 2020-12-02:
how about this in Configuration.h 
```
#ifdef BLTOUCH
  //#define BLTOUCH_FULLY_ON_PROBE_PLUG 
  //#define BLTOUCH_ON_Z_ENDSTOP_PLUG
  //#define BLTOUCH_ON_PIN_27_BOARD
  #if NONE(BLTOUCH_FULLY_ON_PROBE_PLUG, BLTOUCH_ON_Z_ENDSTOP_PLUG, BLTOUCH_ON_PIN_27_BOARD)
    #error "You must select how your BLTOUCH is connected to the controller. Choices are BLTOUCH_FULLY_ON_PROBE_PLUG, BLTOUCH_ON_Z_ENDSTOP_PLUG or BLTOUCH_ON_PIN_27_BOARD."
  #endif
#endif
```
With appropriate changes elsewhere so it functions as expected.

mgrrx, 2020-12-04:
I'm about to mount a BLTouch on my new Ender 5 Pro v 4.2.2 Mainboard. @SlackHacky you mentioned that you chose option 1 and self compiled the firmware, right? Which config did you use?

Skiddz, 2021-01-28:
> 
> 
> I use option 1 when I self compile, this is not a BUG as such, more no clear documentation for a User that did not know these configurations.

I just got my SKR E3 Mini V2 working in my CR-10 Mini, but had to use the Z-stop port for my BL Touch 3.1.  I *think* I know what I need to change in configuration.h to be able to use all 5 wires in the probe port, but can I get a sanity check before I compile and pull it all apart again?  This is my 1st foray into Marlin so I'm still learning.

Comment out #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
Uncomment //#define USE_PROBE_FOR_Z_HOMING
Uncomment //#define Z_MIN_PROBE_PIN 32 (I assume Pin 32 is the Zmin pin in the probe header?)

zaped212, 2021-08-04:
For reference I just received a new ender5 pro yesterday.
I was able to use the configuration files found here:
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-5%20Pro/CrealityV422

but I did need to modify:
#define Z_MIN_PROBE_PIN PB1
to use the 5 pin connector for my BLTOUCH. Finding the correct port was the longest part of building the firmware myself.

It would be nice if we had that as a default in the code atleast with a comment block explaining when to turn it on.

ellensp, 2021-08-04:
default is to use zmin pin ie in the Configuration.h is #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN. 

If you want to use the probe port you disable #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
and enable #define USE_PROBE_FOR_Z_HOMING

and it use the probe port automatically. 
from file Marlin/src/pins/stm32f1/pins_CREALITY_V4.h
```CPP
#ifndef Z_MIN_PROBE_PIN
  #define Z_MIN_PROBE_PIN                   PB1   // BLTouch IN
#endif
```

ellensp, 2021-08-04:
Also if you have the pin 27 board, which plugs the servo into the lcd connector pin, you add #define HAS_PIN_27_BOARD  to your Config file.  (only for boards V4.2.2. V4.2.7)

closed 2021-08-04 by ellensp

zaped212, 2021-08-04:
@ellensp thanks for the info, I will adjust my configuration to that.

I initially had both Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and USE_PROBE_FOR_Z_HOMING defined, and looked like it was trying to use the probe for homing but was expecting it to be on the wrong pin.

I never did try without the Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN, except for when I changed Z_MIN_PROBE_PIN. Thanks!