MarlinFirmware/Configurations #356: Request: Eryone Thinker SE on Marlin 2.0.7.2

bryanpedini, 2020-12-02:
Eryone made available their configs for v2.0.x over [here](https://github.com/Eryone/mega2560), tho they fail to compile if put under 2.0.7.2 firmware.

Please take this printer into consideration too, I'm trying to enable manual mesh bed leveling and the latest 2.0 features, but I don't trust 2.0.x (probably beta) they provided enough to use it in "production".

Thank you very much in advance!!

ellensp, 2020-12-02:
[Eryone Thinker V2.7.2 Configs.zip](https://github.com/MarlinFirmware/Configurations/files/5632765/Eryone.Thinker.V2.7.2.Configs.zip)
Stock config updated to v2.7.2 standards.
Compiles
I have no way to test this at all. use caution when testing.

bryanpedini, 2020-12-04:
Thank you so much @ellensp, the other day I uploaded 2.0.7.2 (via SPI, 'cause I get error when flasing bootloaders with my TinyUSB for some reason) by copying over the important configurations (board, volume, etc) from the config provided by Eryone; and the knob didn't turn.
So I guess I'll backup the current configs on another desktop folder, and try the new one to let you know then.

ellensp, 2020-12-18:
Any updates on this?

bryanpedini, 2020-12-18:
Hello,Thank you for the reminder, I'm really sorry for making you wait,Tho I have my bedroom upside down, switching desks and moving furniture, so both printer are (sadly!) currently offline.I'll update ASAP, again, sorry for the wait!

ellensp, 2020-12-29:
closing due to lack of activity

closed 2020-12-29 by ellensp

bryanpedini, 2021-01-07:
Finally got to try this out (again, sorry for the delay, I had the printer in a box 20cm under a roof for **weeks** 😞), tho unfortunately with the current setup and configuration it never moves from the completely blue screen at startup (doesn't even display the logo).
I'll try out various LCD config and stuff to see if at least I get a menu working (didn't even check what the config actually was) and update with the results.

bryanpedini, 2021-01-07:
Little update:
I've ported the 2.0.x config from their official repo to the new 2.0.7.2 additions (what I could find as difference by having the two files open side by side); can't turn the knob, I get the boot screen, I get the EEPROM error after upgrade, tho can't turn the stupid knob thing 😭
[configurations plain+upgrade (stock - no bltouch - no TMC2209).zip](https://github.com/MarlinFirmware/Configurations/files/5784262/configurations.plain%2Bupgrade.stock.-.no.bltouch.-.no.TMC2209.zip)

Recompiling their firmware+configs (the one I also had before, exactly that 2.0.x I said earlier) works fine again.

Needs more investigating on this printer (which flashing a bootloader fails)!!

bryanpedini, 2021-01-07:
Other update:
WTF bro? Okay I got a lost cause with this printer.

After third try (first two were to discover that I couldn't get anything past blue screen at all), well, third try was the good one and your provided config landed to the EEPROM error after update.
Still, stuck knob turning...

bryanpedini, 2021-01-09:
problem found, configuration above provided works with new board pins provided by Eryone.
this printer is more and more becoming all a mistery...

anyway, thank you a lot @ellensp!! issue can stay closed

PR #395 (2021-01-11): added Eryone Thinker SE based on Thinker V2 board (stock)
source: https://github.com/Eryone/mega2560 + issue #356
with code courtesy of #ellensp

### Requirements

I think none

### Description

Brought Eryone's configuration with the much appreciate help of @ellensp and a bit of trial and error to the Thinker SE printer (on firmware configuration version: 2.0.7.2)

### Benefits

New printer configuration examples: Eryone Thinker SE with board Thinker V2

### Related Issues

feature request / definitively fixes #356