MarlinFirmware/Configurations #358: restore mesh after G28

Spelth, 2020-12-04:
When i do G28 in startup commands :
"
* Startup commands
 *
 * Execute certain G-code commands immediately after power-on.
 */
#define STARTUP_COMMANDS "G28\nM420 S1"
"

And with my gcode start :
"
;Preheat
M104 S210
M109 S210

;start song
M300 S2637 P150
M300 S3135 P150
M300 S5274 P150
M300 S4186 P150
M300 S4698 P150
M300 S6271 P150

;gcode start
G28                   ; home
G90                   ; absolute pos
M82                   ; E Absolute
G92 E0                ; reset pos
G1 X5 Y5 Z0.2 F1000   ; move to bed edge
G1 X15 E15 F500       ; prime extruder
G1 X20 F6000          ; wipe
G1 Y20 E15 F6000      ; wipe and lift
"
the mesh of my bed levelling Bi linear  is not restore for the print.

ellensp, 2020-12-04:
As fas as I know Bilinear probe points are not saved anywhere.
You need to re-probe with G29 every time. 
Leveling is disabled if do a G28 asfter a G29. unless you have enabled RESTORE_LEVELING_AFTER_G28
It has nothing to to wit reloading probing data 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-12-04 by ellensp

thinkyhead, 2020-12-06:
> the mesh of my bed levelling Bi linear is not restore for the print

If you do `M500` after `G29` the mesh is stored to EEPROM. You can verify this with `M503`.

`RESTORE_LEVELING_AFTER_G28` causes leveling to be enabled only if it was enabled _before_ `G28`. You should save your mesh with `M500` as mentioned, and at the start of your G-code you should put `M420 S1` after the `G28` for compatibility with most setups.

Spelth, 2020-12-07:
> > the mesh of my bed levelling Bi linear is not restore for the print
> 
> If you do `M500` after `G29` the mesh is stored to EEPROM. You can verify this with `M503`.
> 
> `RESTORE_LEVELING_AFTER_G28` causes leveling to be enabled only if it was enabled _before_ `G28`. You should save your mesh with `M500` as mentioned, and at the start of your G-code you should put `M420 S1` (to enable leveling with the mesh currently in RAM) after the `G28` for best compatibility with most setups.

Yes is what i do now, and yes it work since i add M420 S1 in my start gcode after G28. 

But normaly with G28, is auto... ,usually, just with G28 i have with M420 V always Bilinear on . But since last update i have not.

thinkyhead, 2021-04-15:
We now have `ENABLE_LEVELING_AFTER_G28`.