MarlinFirmware/Configurations #361: CR-10 Max Config

christianitus03, 2020-12-08:
I would like to upgrade my CR-10 Max's firmware, but there is no configuration for it. What should I do? Can someone smarter than me create one, or at least point me in the right direction? Thank you.

ellensp, 2020-12-08:
I believe there are issues supporting that printers screen in standard Marlin at this time.

InsanityAutomation has a fork of marlin that supports it https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0

christianitus03, 2020-12-08:
> I believe there are issues supporting that printers screen in standard Marlin at this time.
> 
> InsanityAutomation has a fork of marlin that supports it https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0

Yes, I just tried that. Unfortuneatly it appears that the configuration is out of date: compiling gives me "Please update your configuration." errors. I posted an issue there.

ellensp, 2020-12-08:
They really want to you use their binaries...

christianitus03, 2020-12-10:
> They really want to you use their binaries...

Who and what are you referring to?

ellensp, 2020-12-10:
Insanity Automation see all the premade firmwares https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0/Hex%20Files
The key to what all those names means in on the front page https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0

christianitus03, 2020-12-10:
> Insanity Automation see all the premade firmwares https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0/Hex%20Files
> The key to what all those names means in on the front page https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0

Okay, but what do I do with the hex files?

JustinSatu, 2021-01-25:
I'm looking for the same configuration files (Configuration.h, Configuration_adv.h, etc...)
I don't want to work with hex files, because I prefer to be able to edit/finetune settings for my CR-10 Max.

patnodus, 2021-05-13:
Having the same issue

patnodus, 2021-05-13:
> I'm looking for the same configuration files (Configuration.h, Configuration_adv.h, etc...)
> I don't want to work with hex files, because I prefer to be able to edit/finetune settings for my CR-10 Max.

Did you find what you were looking for? @JustinSatu

StudioRATATA, 2021-08-13:
+1 
Does anyone have a solution or even the configs to share?

patnodus, 2021-08-13:
> +1
> Does anyone have a solution or even the configs to share?

I haven't found anything of this yet.  Sorry

Tarsonoff, 2022-01-05:
Я так же нечего не нашёл! Если найдёте прикрепите пожалуйста.

thinkyhead, 2022-01-26:
Duplicate of #649

closed 2022-01-26 by thinkyhead