MarlinFirmware/Configurations #365: BL-Touch Fails To Move

omarhabib11, 2020-12-10:
### Bug Description

Hello. The Creality CR-10 V2 configuration with BL-Touch fails to probe successfully. It will probe as expected along the first row, but when it gets to the second row along the Y axis, it proceeds to probe the same spot multiple times. The printer reports the X axis as moving properly for the subsequent probes, although the probe stays in the same spot, until it moves down another row along the Y axis, thinking it is at the reported X axis. It thinks it is near X=0, but actually, X is almost at the end because it never moved backwards, so it tries to move past the end of the X axis for the next probes, causing the stepper to jam and make an awful noise.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Start Bed Levelling
2. Wait for failure around probe 6 or 7

**Expected behavior:**

Probing should not cause the stepper to jam.

**Actual behavior:**

Stepper jams on the X axis after a few probes.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

dirk-schulz, 2020-12-11:
Hello, 
I have an Artiellery Sidewinder X1 and I can confirm to have the same stepper jam with BLTouch leveling with this firmware version 2.0.7.2. 
When this issue occurs, the printer "thinks" that x=0 and y=0 is in the middle of the bed. So even if you try to move it manual via TFT screen, the printer head does not move more to the left as the middle of the print bed seems to be the x home position. 
At the moment I only can solve that by turn off and turn on the printer.

Any fix for that? Does the 2.0.5.x version has the same behaviour?

ellensp, 2020-12-12:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

dirk-schulz, 2020-12-13:
Hello, I adopted my config for the sidewinder in the provided bugfix-2.0.x branch. But I got errors while compiling.
Seems that in the bugfix files soming is going wrong?

Compiling .pio\build\mega2560\FrameworkArduino\wiring_pulse.S.o
Compiling .pio\build\mega2560\FrameworkArduino\wiring_pulse.c.o
C:\users\disc\.platformio\packages\framework-arduino-avr\libraries\Wire\src\utility\twi.c: In function '__vector_39':
C:\users\disc\.platformio\packages\framework-arduino-avr\libraries\Wire\src\utility\twi.c:552:7: warning: empty declaration
       __attribute__ ((fallthrough));
Compiling .pio\build\mega2560\FrameworkArduino\wiring_shift.c.o
       ^
C:\users\disc\.platformio\packages\framework-arduino-avr\libraries\Wire\src\utility\twi.c:638:7: warning: empty declaration
       __attribute__ ((fallthrough));
       ^
Archiving .pio\build\mega2560\lib4e0\libWire.a
Archiving .pio\build\mega2560\libFrameworkArduino.a
Linking .pio\build\mega2560\firmware.elf
.pio\build\mega2560\src\src\lcd\ultralcd.cpp.o (symbol from plugin): In function `MarlinUI::set_status(char const*, bool)':
(.text+0x0): multiple definition of `MarlinUI::set_status(char const*, bool)'
.pio\build\mega2560\src\src\lcd\marlinui.cpp.o (symbol from plugin):(.text+0x0): first defined here
.pio\build\mega2560\src\src\lcd\ultralcd.cpp.o (symbol from plugin): In function `MarlinUI::set_status(char const*, bool)':
(.text+0x0): multiple definition of `MarlinUI::set_status_P(char const*, signed char)'
.pio\build\mega2560\src\src\lcd\marlinui.cpp.o (symbol from plugin):(.text+0x0): first defined here
.pio\build\mega2560\src\src\lcd\ultralcd.cpp.o (symbol from plugin): In function `MarlinUI::set_status(char const*, bool)':
(.text+0x0): multiple definition of `MarlinUI::status_printf_P(unsigned char, char const*, ...)'
.pio\build\mega2560\src\src\lcd\marlinui.cpp.o (symbol from plugin):(.text+0x0): first defined here
.pio\build\mega2560\src\src\lcd\ultralcd.cpp.o (symbol from plugin): In function `MarlinUI::set_status(char const*, bool)':
(.text+0x0): multiple definition of `MarlinUI::eeprom_alert(unsigned char)'
.pio\build\mega2560\src\src\lcd\marlinui.cpp.o (symbol from plugin):(.text+0x0): first defined here
(.text+0x0): multiple definition of `ui'
.pio\build\mega2560\src\src\lcd\marlinui.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\mega2560\firmware.elf] Error 1
================================================================================================== [FAILED] Took 36.17 seconds ==================================================================================================
Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:00:36.165
============================================================================================= 1 failed, 0 succeeded in 00:00:36.165 =============================================================================================

dirk-schulz, 2020-12-13:
Hello,
as a first feedback I had to report, that during compiling I got the following issue with twi as output.
The compiling at the end was successful. Can I still upload this firmware to the printer?

Here the isse during compiling:
Compiling .pio\build\mega2560\FrameworkArduino\new.cpp.o
C:\users\disc\.platformio\packages\framework-arduino-avr\libraries\Wire\src\utility\twi.c: In function '__vector_39':
C:\users\disc\.platformio\packages\framework-arduino-avr\libraries\Wire\src\utility\twi.c:552:7: warning: empty declaration
       __attribute__ ((fallthrough));
Compiling .pio\build\mega2560\FrameworkArduino\wiring.c.o
       ^
C:\users\disc\.platformio\packages\framework-arduino-avr\libraries\Wire\src\utility\twi.c:638:7: warning: empty declaration
       __attribute__ ((fallthrough));    
       ^
Compiling .pio\build\mega2560\FrameworkArduino\wiring_analog.c.o
Compiling .pio\build\mega2560\FrameworkArduino\wiring_digital.c.o

dirk-schulz, 2020-12-13:
Update: The upload to the printer worked well. I had to customize the configuration.h and configuration_adv.h for my sidewinder X1 first. (The issue above is still shown while compiling). 
I will now make some prints to test, if the stepper jam still occurs or not. I will keep you updated.

dirk-schulz, 2020-12-13:
Sorry guys,
for me the issue still exists also with the bugfix from today! Sporadic (so not every time) while doing the probe of the ABL, the printer head is no longer moving on the x axis and stay on the same position. The display is showing that x is changing, but in real the print head is not leaving it's position and probes on the same position. Once the probe of the current row are done, y axis is moving to the next row and then X axis is moving from it's position to the end of the x axis and crashs with the end of the x axis. (printer head is not on the x position as the display is assuming it is) 
In some cases this behavior comes with the error message that there was a probe issue, but the probe procedure is not stopping and still continue (to crash / stepper jam).
Also with manual moving the x-axis, the printer head is only moving as left as were the probe issue was. But the display still shows an decreasing x, even if the printer head itself is no longer moving.
Also the endstops switches (Y & X) are no longer working. Moving along Y the endstop switch is ignored and trying to move further.
It can only be fixed by switch off and on the printer.
Please investigate further the issue

omarhabib11, 2020-12-13:
I am experiencing the same issue as dirk-schulz with the latest bugfix branch. Seems like the issue might be deeper than config.