MarlinFirmware/Configurations #372: Tevo/Homers Tarantula Pro unable to compile firmware.

Circomedia, 2020-12-23:
Hello, I am having issues building the firmware in the Auto Builder. 

I have searched high and low for the solution to updating the firmware and I can not find a solution anywhere. 

`Windows PowerShell

PS D:\Desktop\Marlin-2.0.x> platformio run -e mega2560
Processing mega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.0.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0 
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.3
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0
Building in release mode
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\eeprom.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fast_pwm.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fastio.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\watchdog.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\HAL_spi_L6470.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_if_i2c.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_if_spi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\mega2560\src\src\MarlinCore.cpp.o
Compiling .pio\build\mega2560\src\src\core\serial.cpp.o
Compiling .pio\build\mega2560\src\src\core\utility.cpp.o
Compiling .pio\build\mega2560\src\src\feature\bedlevel\bedlevel.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\calibrate\G28.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M200-M205.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M220.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M221.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M301.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M302.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M92.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M108_M112_M410.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M111.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M120_M121.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M17_M18_M84.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M211.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M280.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M350_M351.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M80_M81.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M85.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M993_M994.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M997.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M999.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\T.cpp.o
[... 112 lines cut ...]
Marlin\src\module\../inc/../core/macros.h:196:29: note: in expansion of macro 'THIRD'
 #define ___TERN(P,V...)     THIRD(P,V)              // If first argument has a comma, A. Else B.
                             ^
Marlin\src\module\../inc/../core/macros.h:195:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^
Marlin\src\module\../inc/../core/macros.h:194:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
Marlin\src\module\../inc/../core/macros.h:190:29: note: in expansion of macro '_TERN'
 #define TERN(O,A,B)         _TERN(_ENA_1(O),B,A)    // OPTION converted to '0' or '1'
                             ^
Marlin\src\module\motion.cpp:513:60: note: in expansion of macro 'TERN'
   do_blocking_move_to_z(_MIN(zdest, Z_MAX_POS), MMM_TO_MMS(TERN(HAS_BED_PROBE, Z_PROBE_SPEED_FAST, HOMING_FEEDRATE_Z)));
                                                            ^
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
*** [.pio\build\mega2560\src\src\module\motion.cpp.o] Error 1
====================================================================== [FAILED] Took 22.84 seconds ====================================================================== 

Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:00:22.837
================================================================= 1 failed, 0 succeeded in 00:00:22.837 ================================================================= 
PS D:\Desktop\Marlin-2.0.x> echo "done" >C:\Users\Angus\AppData\Local\Temp\ipc
PS D:\Desktop\Marlin-2.0.x> I`

Circomedia, 2020-12-23:
"void do_z_clearance(const float &zclear, const bool z_known/*=true*/, const bool raise_on_unknown/*=true*/, const bool lower_allowed/*=false*/) {
  const bool rel = raise_on_unknown && !z_known;
  float zdest = zclear + (rel ? current_position.z : 0.0f);
  if (!lower_allowed) NOLESS(zdest, current_position.z);
  do_blocking_move_to_z(_MIN(zdest, Z_MAX_POS), MMM_TO_MMS(TERN(HAS_BED_PROBE, Z_PROBE_SPEED_FAST, HOMING_FEEDRATE_Z)));
}" should it be "MMM" ?

ellensp, 2020-12-23:
Your configuration files your using is newer than the code you are using.

The code is looking for HOMING_FEEDRATE_XY and HOMING_FEEDRATE_Z
but this config has the newer HOMING_FEEDRATE_MM_M

Find the line #define HOMING_FEEDRATE_MM_M in your Configuration.h
eg #define HOMING_FEEDRATE_MM_M { (50*60), (50*60), (4*60) }

add the lines #define HOMING_FEEDRATE_XY (50*60) and #define HOMING_FEEDRATE_Z (4*60)
Where the number are the X and Z feed rates from HOMING_FEEDRATE_MM_M

closed 2020-12-23 by ellensp