MarlinFirmware/Configurations #376: [BUG] (short description)

legend069, 2020-12-25:
### Bug Description

missing "#define PROBE_OFFSET_START -4.0" in the function for 
  "   
//#define PROBE_OFFSET_WIZARD
    #if ENABLED(PROBE_OFFSET_WIZARD) 
    "
and

#if EITHER(ULTIPANEL, EXTENSIBLE_UI)
  #define MANUAL_FEEDRATE { 50*60, 50*60, 4*60, 2*60 } // (mm/min) Feedrates for manual moves along X, Y, Z, E from panel
  #define SHORT_MANUAL_Z_MOVE 0.025 // (mm) Smallest manual Z move (< 0.1mm)
  #if ENABLED(ULTIPANEL)
    #define MANUAL_E_MOVES_RELATIVE // Display extruder move distance rather than "position"
    #define ULTIPANEL_FEEDMULTIPLY  // Encoder sets the feedrate multiplier on the Status Screen
  #endif
#endif

"SHORT_MANUAL_Z_MOVE" - correct 
"FINE_MANUAL_MOVE" - incorrect.

prevents compiling.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

**Actual behavior:**

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2020-12-25:
no.
#error "SHORT_MANUAL_Z_MOVE is now FINE_MANUAL_MOVE, applying to Z on most printers."

ellensp, 2020-12-25:
Didn't follow template, seem to be nonsense. Closing till follows template and config files are provided.

closed 2020-12-25 by ellensp

legend069, 2020-12-25:
it seems like the default config file for the "D:\Downloads\temp marlin\Configurations-import-2.0.x(1)\Configurations-import-2.0.x\config\examples\Creality\Ender-3\BigTreeTech SKR Mini E3 2.0" is all wrong.

ellensp, 2020-12-25:
Import branch is not for users. It part of the configuration generation system.
Download the configuration for the version of Marlin you are using. 
Ie your using marlin 2.0.x so use latest release branch release-2.0.7.2

ellensp, 2020-12-25:
ie like the readme in the 2.0.x/config directory says to do
"Where have all the configurations gone?
https://github.com/MarlinFirmware/Configurations/archive/release-2.0.7.2.zip"

legend069, 2020-12-25:
ohh i see where i went wrong the "import branch" is the default tree at "https://github.com/MarlinFirmware/Configurations"