MarlinFirmware/Configurations #378: "BigTreeTech SKR Mini E3 2.0", HOMING_FEEDRATE_XY missing in config.

Cyberman-tM, 2020-12-26:
When copying the configuration into the latest marlin source, I'm greeted with this error upon compiling:

`Compiling .pio\build\STM32F103RC_btt_512K_USB\src\src\module\stepper\trinamic.cpp.o
In file included from Marlin\src\module\../inc/MarlinConfig.h:47:0,
                 from Marlin\src\module\motion.h:31,
                 from Marlin\src\module\motion.cpp:27:
Marlin\src\module\motion.cpp:151:16: error: 'HOMING_FEEDRATE_XY' was not declared in this scope
     MMM_TO_MMS(HOMING_FEEDRATE_XY), MMM_TO_MMS(HOMING_FEEDRATE_XY),
                ^
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
                                           ^~~~
Marlin\src\module\motion.cpp:151:16: note: suggested alternative: 'HOMING_FEEDRATE_MM_M'
     MMM_TO_MMS(HOMING_FEEDRATE_XY), MMM_TO_MMS(HOMING_FEEDRATE_XY),`
(I didn't change anything in the config yet.)

The problem is that for some reason the HOMING_FEEDRATE_XY was removed in this configuration, and replace (?) with 
`#define HOMING_FEEDRATE_MM_M { (20*60), (20*60), (4*60) }`

What I did then is add these lines:
`#define HOMING_FEEDRATE_XY (50*60)
#define HOMING_FEEDRATE_Z  (4*60)`

Now it compiles. I have NOT actually tried it yet!

I'd make a pull request, but I'm not sure if this is really a solution or not - but I didn't find anything useful otherwise, so I'm opening this up. I'm probably not the only one having that problem?

thisiskeithb, 2020-12-26:
Your config version needs to match the version of Marlin you're compiling for. `HOMING_FEEDRATE_XY` was replaced by `HOMING_FEEDRATE_MM_M`, so configs from `bugfix-2.0.x`/`import-2.0.x` are only compatible with Marlin `bugfix-2.0x`.

Use [Marlin 2.0.7.2](https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.7.2) with the [2.0.7.2 configs](https://github.com/MarlinFirmware/Configurations/tree/release-2.0.7.2) or move to bugfix-2.0.x for both.

closed 2020-12-26 by thisiskeithb

Cyberman-tM, 2020-12-26:
That makes sense, of course.

What doesn't make sense (to me at least), is why the repositories aren't on the level, then?
I downloaded the default branch from Marlin and Configuration, assuming they'd be equal.
Apparently configuration is on bugfix by default? Would "Master" be the same as marlin?

thisiskeithb, 2020-12-26:
> I downloaded the default branch from Marlin and Configuration, assuming they'd be equal.

That won't work as you've found out.

> Apparently configuration is on bugfix by default? Would "Master" be the same as marlin?

You just have to pay attention to which branch/release you're on. Use [Marlin `2.0.7.2`](https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.7.2) with the [`2.0.7.2` configs](https://github.com/MarlinFirmware/Configurations/tree/release-2.0.7.2) or move to[ Marlin `bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) + [`bugfix-2.0.x` configs](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/)

If it's easier, the [Marlin Firmware Downloads](https://marlinfw.org/meta/download/) page has versions paired for you.