MarlinFirmware/Configurations #390: [BUG] Tronxy Chitu_V6 TFT not working right

NicholasTracy, 2021-01-06:
### Bug Description
After flashing to the recent bugfix version of marlin 2.0.x and using the new configuration files example for the X5SA, The TFT screen does not work as expected. Touch screen is off scale.

### Configuration Files
[Configuration.zip](https://github.com/MarlinFirmware/Configurations/files/5774975/Configuration.zip)

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

I Tried using the "out of the box" configuration files provided in the example folder. But the screen shows up in the top left corner smaller than the rest of the screen. Default resolution of (320x240) seems to be wrong. After changing this to (480x320) it displays correctly, however the touch screen interface is way out of whack. It seem that the top left buttons are in the correct spot but the far bottom right buttons on screen do not work at all. middle of the screen seem slightly to the right and down from where it should be. 

#### Additional Information
I have tried it with the Classic UI option and the Color UI option. Neither worked.
I had a previous version from around a few months ago that worked just fine but the configuration file simply had an option for FSMC TFT. The recent version has a bunch of new options including resolution and UI that wasn't there before. Tried forcing a screen calibration with M995 in auto0.g file on SD card but the calibration always fails. 

I can try tomorrow to get video of the issue in action.

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

krukhlis, 2021-01-19:
Hi there. For me it's the same with my X5SA, it shows small screen in the top left corner, but touch screen operates in "full-screen mode". If I change to 480x320 -- it looks in full screen mode, but touch screen stops working completely.
But I have found solution( at least for me).
1. Change touch screen defines( ~ line 2461) to:
 #define TOUCH_SCREEN_CALIBRATION

  //#define TOUCH_CALIBRATION_X -12316
  //#define TOUCH_CALIBRATION_Y   8981
  //#define TOUCH_OFFSET_X         340
  //#define TOUCH_OFFSET_Y         -20
  #define TOUCH_ORIENTATION TOUCH_LANDSCAPE
 2. Compile and flash it to your tronxy.
 3. Connect your laptop/pc using USB cable to motherboard of your printer and issue through the terminal the following commands:
 a) M995; this will trigger touchpad calibration on your screen.
                Follow screen calibration procedure( around 5 mins clicking with your finger here and there). First 2 rounds it says failed and then it says "SUCCESS". After successful confirmation on your printer's screen, type the following command in your terminal
 b) M500; to store your settings               
 4. Disconnect your laptop/PC
 5. Enjoy fully functional full-screen Marlin 2 touch-controlled screen. :)
 P.S. And I have Classic UI option.

gb0101010101, 2021-01-20:
I have X5SA V5 (Board ID: CXY-V5-180409) and screen is displayed at correct size but touch calibration seems to be reversed. 
I tried `#define TFT_GENERIC` and `#define TFT_TRONXY_X5SA`. Both settings displayed at correct size but had unusable touch calibration. 

Was able to send gcode M995 and successfully calibrate on first try. My touch screen config parameters are: 
```
#define TOUCH_CALIBRATION_X -17217
#define TOUCH_CALIBRATION_Y 11262
#define TOUCH_OFFSET_X 501
#define TOUCH_OFFSET_Y -14
#define TOUCH_ORIENTATION TOUCH_LANDSCAPE
```
Looks like there are variations between board IDs. Perhaps config examples  
should be split out by Board ID? e.g. 
TRONXY_X5SA_V5
TRONXY_X5SA_V6

NicholasTracy, 2021-01-20:
Agreed, I have noticed that the current examples don't make a ton of sense for my v6 board. My X5SA non pro and without the titan extruder comes with the V6 board and has the TMC stand alone drivers. yet the examples make it seem like only the pro version has this with a titan extruder. On top of the issues the the TFT calibration (which I will be testing out tonight to see if either of these solutions work for my case.) it seems like there could be some improvements made to the config file. I will likely create my own fork and split out the configs by board revision.

gb0101010101, 2021-01-21:
I have X5SA V5 (Board ID: CXY-V5-180409) and screen is displayed at correct size but touch calibration seems to be reversed and smaller. I tried `#define TFT_GENERIC` and `#define TFT_TRONXY_X5SA`. Both settings displayed at correct size but had unusable touch calibration. 

Was able to send gcode M995 and successfully calibrate on first try. Note non-standard serial baud rate of 250000.
My touch screen config parameters are: 
```
#define TOUCH_CALIBRATION_X -17217
#define TOUCH_CALIBRATION_Y 11262
#define TOUCH_OFFSET_X 501
#define TOUCH_OFFSET_Y -14
#define TOUCH_ORIENTATION TOUCH_LANDSCAPE
```
Looks like there are variations between board IDs. Perhaps config examples should be split out by Board ID? e.g. 

- TRONXY_X5SA_V5
- TRONXY_X5SA_V6

gb0101010101, 2021-02-08:
TronXY seems to have shipped out a number of different variants of the XS5A that are not the "Pro" version but are differing from the "Basic" spec. When I bought the printer on Amazon (always take a screenshot), the page said "adopts German imported TMC ultra-quiet motor drive chip" and "Materials support: PLA, ABS, HIPS, PETG, WOOD, PC, PVC ". Neither of these were true for the machine I received. It had A4988 chips on TronXY v1.5 board and used the "basic" extruder [read: not Titan clone] that only officially supported PLA and ABS. Thankfully Amazon agreed to a refund, when I said this was "Bait and switch", with pre-paid [PRIME] return shipping, even after the 1 month return interval.

AFAIK only TronXY v1.8 boards onward use TMC chips. Inclusion of "Titan clone" extruder that supports [PLA, ABS, HIPS, PETG, WOOD, PC, PVC] still seems random.

Would love to hear specs from other users to establish a baseline but it seems like a small customer base using this possibly fine controller board.

krukhlis, 2021-02-08:
I've got my tronxy X5SA from Canadian Amazon: https://www.amazon.ca/gp/product/B087NB7S9Q/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
It's regular one, but revision of 2020 year. Majority of issues mentioned by people that purchased X5SA in previous years are already resolved.
The stock board is STM32-based with TMC2225 drivers( I believe it's 1.8 revision). Flashed without any issues with Marlin v2( beside the need to recalibrate touch and increase resolution as I've mentioned in my comment https://github.com/MarlinFirmware/Configurations/issues/390#issuecomment-762540061 ).

NicholasTracy, 2021-02-08:
So after further tinkering with the config file using suggestions from above, I am able to get it all working. I commented out all the preset calibration numbers and uncommented touch_orientation touch_landscape. This seems to have solved the issue with the TFT touch screen after doing a calibration. However, now it crashes the Z when homing. It seems as though the firmware isn't recognizing the endstop trigger even though the inductive sensor is triggering. So for now I had to revert back to the latest Marling release. I haven't been able to figure out why it won't trigger. The sensor is definitely sending the correct signals, the config file has the correct settings for the z probe. When using the latest release version it works fine. Its only happening with the bugfix version. If anyone has any insight in to that, I'd appreciate it. I want to be able to use all the new features and the new TFT GUI options. The latest stable release doesn't have any of that.

krukhlis, 2021-02-08:
> Inclusion of "Titan clone" extruder that supports [PLA, ABS, HIPS, PETG, WOOD, PC, PVC] still seems random.

Well, I don't care about stock feeder, because both stock feeder and stock hotend are crappy( the samy crappy feader was installed on my Ender 3). Regardless to say that usage of bowden for such large printers is a crime and stupid engineering. I have replaced them with direct drive with BMG extruder and slim motor + Micro Swiss All metal hotend. Works like a charm and I can print any plastic.

NicholasTracy, 2021-02-08:
It would seem to me at least that adding a line and comment for either option would be uncomplicated and trivial. That way people looking through the config file can easily find the option for either or. Same goes for the steps per parameters... Since I have the TMC drivers on my ChituV6 board I find myself having to always make sure those are changed when I start a new config file. It should have the standard parameters for both drivers readily available as an option with one of them being commented out as default.

gb0101010101, 2021-02-10:
@NicholasTracy 
> However, now it crashes the Z when homing

Has the NOZZLE_TO_PROBE_OFFSET values changed? The third value is Z I set mine as 0 and adjust afterwards using UI Z offset. Otherwise might be a pin setting Z_MIN_PIN or Z_MIN_PROBE_PIN. These values can get renamed between firmware versions so config files need updating too.

NicholasTracy, 2021-02-13:
> @NicholasTracy
>
>
> Has the NOZZLE_TO_PROBE_OFFSET values changed? The third value is Z I set mine as 0 and adjust afterwards using UI Z offset. Otherwise might be a pin setting Z_MIN_PIN or [...]

Mine is set to 0 as well which is the default for the config example. Everything in the config file checks out to be correct. I have not checked to see if there is an issue with the pins definition, if that isn't it then I am going to assume its just a bug in Marlin with that bugfix version. I'm going try the most recent bugfix tonight and see if it does it again.

mrl22, 2021-06-01:
I have the latest 2021 Tronxy X5SA Pro with TMC, Titan Extruder etc. I have managed to compile and ran into similar issues as above. To fix the screen size issue I commented out #define TFT_RES_320x240 and enabled #define TFT_RES_480x320. After this, the touch did not work. I created a file called auto0.g on the sd card which contains M995; This file is automatically run on startup and allowed me to calibrate the touch.

The printer and screen are now working perfectly.

What I would like to know is how I would go about downloading my saved touch settings so that I can share and write into the firmware? As @gb0101010101 did.

Thank you for sharing, this thread had helped a lot.