MarlinFirmware/Configurations #393: M0 and M117 aren't working on Creality printers with TFT screens

GregValiant, 2021-01-07:
### Bug Description
I don't know that this could be called a bug.  I was hoping for an explanation though.
I am active on the Ultimaker Cura forum and on Reddit.  Lately there have been complaints that the Cura PauseAtHeight plugin is not working on some Creality printers.  There have also been some reports that M117 is not working on some Creality printers.  I think both problems are related because Cura uses M0 as the pause command and M0 sends a message to the screen and of course that is the purpose of M117.
I have no config.h or any other documentation nor have I heard of this problem effecting other printers.  One poster stated that he was able to get M0 to work by "...TFT to Bigtreetech TFT35 V3 as well as my board to a SKR 1.4 Turbo and had some very smart people tweak my Marlin".
So my question is, is there a tweak for the stock Creality firmware that will allow M0, M117 to work with a TFT screen?  I have the older style LCD and there is no problem pausing or sending messages to the LCD.

### Links to some posts:

Here [https://www.reddit.com/r/ender3/comments/kmc56y/did_i_understand_this_right_ender_3_v2_cant_show/](url)
and here [https://community.ultimaker.com/topic/30009-pause-at-height-cura-43-not-working/?tab=comments#comment-280377](url)

### Steps to Reproduce

Sending an M117 command does nothing
Sending an M0 or M1 command does nothing

**Expected behavior:**

When "M117 Hello" is sent to the printer via USB then "Hello" should show up on the printer display.

When M0 or M1 is sent to the printer via USB the printer should pause and "Click to Continue" should appear on the printer display.

#### Additional Information

None

ellensp, 2021-01-08:
1) this is for Marlin example configs issues. you have no configs
2) this is fro actual marlin, not modified marlin that creality uses. 
To help help with creality firmware you have to talk to creality.

closed 2021-01-08 by ellensp

GregValiant, 2021-01-08:
Thanks for your time.