MarlinFirmware/Configurations #400: [BUG] (Encoder wheel not working properly on Ender 3 V2 with latest marlin bugfix and configuration bugfix)

Leo06ba, 2021-01-15:
The encoder wheel on the Ender 3 V2 doesn't work properly. It increases values by 1 and unselects the object on the scree so you have to click it again for every increase. Tested with latest marlin bugfix and configuration bugfix.

ooksanen, 2021-01-15:
Can confirm.

There seems to be a problem with the screen serial port and gives errors ("LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2.") during compiling if SERIAL_PORT_2 value is set to 3 as instructed in Teaching Tech video https://youtu.be/neS7lB7fCww?t=508

Reverting to 2.0.7.2 fixes these problems.

Leo06ba, 2021-01-15:
> Can confirm.
> 
> There seems to be a problem with the screen serial port and compiling errors ("LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2.") if SERIAL_PORT_2 value is set to 3 as instructed in Teaching Tech video https://youtu.be/neS7lB7fCww?t=508
> 
> Reverting to 2.0.7.2 fixes these problems.

Weird thing is that I compiled last week with the previous bugfix release and i had no issues.

ooksanen, 2021-01-15:
Yeah, I had a bugfix branch firmware compiled a few weeks ago too and working without problems.

ooksanen, 2021-01-16:
This is a duplicate of https://github.com/MarlinFirmware/Marlin/issues/20774 and is fixed in https://github.com/MarlinFirmware/Marlin/commit/4472ba2b6bc0835d0fcc039ad66d9dab40e13010

My compiling problem seems to be a separate issue.

ellensp, 2021-01-16:
@ooksanen 
the default lcd serial is serial 3
```
#if ENABLED(DWIN_CREALITY_LCD)
  #define SERIAL_CATCHALL 0
  #ifndef LCD_SERIAL_PORT
    #define LCD_SERIAL_PORT 3
```
So you need to change either LCD_SERIAL_PORT or SERIAL_PORT_2 to be different.

closed 2021-01-16 by ellensp

Leo06ba, 2021-01-16:
> @ooksanen 
> the default lcd serial is serial 3
> ```
> #if ENABLED(DWIN_CREALITY_LCD)
>   #define SERIAL_CATCHALL 0
>   #ifndef LCD_SERIAL_PORT
>     #define LCD_SERIAL_PORT 3
> ```
> So you need to change either LCD_SERIAL_PORT or SERIAL_PORT_2 to be different. 
> 
> If your just trying to enable the LCD with setting SERIAL_PORT_2 to 3, don't. You do not use SERIAL_PORT_2 or SERIAL_PORT for this LCD. You use LCD_SERIAL_PORT which automatically defaults to 3.  
> ie just leave SERIAL_PORT_2 disabled.
> 

So the fix would be to change SERIAL_PORT to 3?

ellensp, 2021-01-16:
no! that would still conflict with LCD_SERIAL_PORT
YOU DO NOT SET A SERIAL PORT FOR THIS LCD YOU LET THE SYSTEM SET #define LCD_SERIAL_PORT 3

Leo06ba, 2021-01-16:
> no! that would still conflict with LCD_SERIAL_PORT
> YOU DO NOT SET A SERIAL PORT FOR THIS LCD YOU LET THE SYSTEM SET #define LCD_SERIAL_PORT 3
> 

Ok, thanks. I think someone fixed the issue in the new marlin bugfix. I definetly will try to compile tomorrow with the new one.