MarlinFirmware/Configurations #401: [FR] BLTouch starts flashing if probe touches something but Z axis stepper motor does not stop

MiMuij, 2021-01-16:
### Description

Hi everyone,

thanks for your great work! I ran my Ender 3 Pro (v4.2.2) on Marlin 2.0.7.2 for a couple of weeks and I like it so much more than the original firmware by Creality!

But now I bought a BLTouch [Amazon link](https://www.amazon.de/gp/product/B08L9DHP5R/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1) for my printer and it's kind of frustrating, that I am not able to get that thing working correctly.

I tried to compile Marlin in the state the source code had two days before here on Github with the [config files provided for the Ender 3 Pro V4.2.2](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3%20Pro/CrealityV422), but I ran into compiling errors, which I weren't able to fix by myself.

So I decided to download the [v2.0.7.2 from the archive](https://github.com/MarlinFirmware/Marlin/archive/2.0.7.2.zip) and also copied the [cofiguration files for "Ender 3 Pro V1.5" from realease 2.0.7.2](https://github.com/MarlinFirmware/Configurations/tree/release-2.0.7.2/config/examples/Creality/Ender-3%20Pro%20V1.5).

I changed the configuration files like shown in these examples:
- [Danieldoesmanythings.com](https://danieldoesmanythings.com/357/configuring-marlin-2-0-for-an-ender-3-pro-and-bltouch/),
- [Marcdobler.com](https://www.marcdobler.com/2020/05/05/configuring-marlin-2-0-for-an-ender-3-pro-and-bltouch/) and
- [This Video](https://youtu.be/YsTGH5JvClM) made by [Crosslink](https://www.youtube.com/channel/UC4M4XC4Bt0jREd4AHGjIymw)

Everything compiled fine! But with Marlin 2.0.7.2 installed, the BLTouch does not work properly: If I hold my finger under the probe while Auto Homing the Z axis the probe gets stowed, the red led on the BLTouch starts to flash but the stepper keeps on moving downwards so that the nozzle will jam into the bed.

I tried to change some of the parameters in the config files, but the behaviour does not change.

I flashed the original firmware from Creality for usage with BLTouch (not the firmware released today) and it worked properly.

What am I doing wrong?

### Configuration Details

* Manufacturer: Creality
* Model: Ender 3 Pro
* Motherboard: v4.2.2

#### Additional Information

* [cofiguration files for "Ender 3 Pro V1.5" from realease 2.0.7.2](https://github.com/MarlinFirmware/Configurations/tree/release-2.0.7.2/config/examples/Creality/Ender-3%20Pro%20V1.5)

ellensp, 2021-01-16:
Not A Feature request. 

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-01-16 by ellensp

MiMuij, 2021-01-17:
Thanks! Have a great day!