MarlinFirmware/Configurations #402: SD card reader disabled on Creality CR-10S

djardi, 2021-01-17:
### Bug Description

The most recent Marlin firmware update disables the SD card reader in Creality CR-10S. This is a configuration issue, not a bug. Having said that, since CR10-S printers come with an SD card slot and reader and since the proper configuration of this setting is rather obscure, I propose that the CR10-S example configuration files should be changed so that the SD card reader is enabled by default. That will save many users a lot of time and frustration trying to use their SD card reader after the update. This is easily accomplished by changing one line of the settings in the configuration_adv.h file:
change
//#define SDCARD_CONNECTION LCD
to 
#define SDCARD_CONNECTION ONBOARD

### Configuration Files
[SD_prob.zip](https://github.com/MarlinFirmware/Marlin/files/5767194/SD_prob.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step]: Download the Marlin 2.0.x firmware for the Creality CR-10S (Configurations-release-2.0.7.2\Configurations-release-2.0.7.2\config\examples\Creality\CR-10S\CrealityV1). Use the configuration files found in the directory. Do NOT change the settings for the SD drive in the configuration_adv.h file.
2. [Second Step]: Try to use the SD card reader. The only menu choice related to the SD card reader is "Attach media," which does nothing. Restarting the printer does not help.

**Expected behavior:**
The menu should show the correct menu options and allow the user to print from the SD card.
[image]
[image]

**Actual behavior:**
There is only one menu choice (Attach media). Selecting this choice does not do anything with the SD card because the SD card reader has not been defined in the setup files. It does not matter whether the printer is turned on with the SD card in the slot or whether the SD card is inserted after the printer is turned on. The printer does not see the SD card. The user is unable to see any files on the SD card. Consequently, printing from the SD card is impossible.
[image]

This is NOT a bug. It is a configuration issue. It seems to me to make more sense to have the default configuration of the configuration_adv.h menu set so that the default SD card reader in CR 10S printers is immediately available to users.

Thanks for considering this change. It would make Marlin 2.0.x firmware much easier to use for Creality CR-10S users.

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

NezSez, 2021-03-16:
"Immediately available to users" can mean either via printer LCD/TFT or via the Computer/Host USB mapped drive but those are mutually exclusive in the current code, even with open/shared serial enabled for host.

On my SKR 1.4 Turbo, bugfix-2.0.x Ender 5 Plus configuration, for either the printer or the Windows10 to list the SD card contents, I have to have at least one actual .gcode file on the card itself, and for my Windows10 USB device to actually mount and pop open an explorer window, which it will only do if I M22/Release SD on the printer first, then Windows10 mounts and displays content immediately.

If I have the card in the SKR but with no .gcode files on it, then the printer will list nothing but still mount the card and show no Print from card options, while still enabling the Release Media option as your screen shows. The Windows10 mount shows the actual drive device but if you click it it says "Please insert" even if I do M22/Release SD.

Maybe I have some other option enabled which causes this, or the code handles things differently on LPC etc, but you might try experimenting and see if this helps you.

thisiskeithb, 2022-04-20:
See https://github.com/MarlinFirmware/Configurations/pull/716

closed 2022-04-20 by thisiskeithb