MarlinFirmware/Configurations #406: Pin Assignment for Mecreator 2 not correct (X & Y Axes)

derstephan2021, 2021-01-25:
Hello!

I build the Marlin 2 Firmware on VS Code successful with Auto Build Marlin. But I ran into problems moving the x and y axis afterwards. So I compared the Pin Numbers from the original Geetech Firmware Source Code with the Marlin Examples and found some differences:

Geetech Pin Assignment:

#define X_STEP_PIN 37 
#define X_DIR_PIN 39
#define X_MIN_PIN 22
#define X_MAX_PIN 24
#define X_ENABLE_PIN 35

#define Y_STEP_PIN 31
#define Y_DIR_PIN 33
#define Y_MIN_PIN 26
#define Y_MAX_PIN 28
#define Y_ENABLE_PIN 29

--------------------------------------------------------------------------------------------

Marlin 2.0 Assignment:

#ifndef BOARD_INFO_NAME
  #define BOARD_INFO_NAME "GT2560 V3.0"
#endif

//
// Servos
//
#define SERVO0_PIN                            11  //13 untested  3Dtouch

//
// Limit Switches
//
#ifndef X_STOP_PIN
  #ifndef X_MIN_PIN
    #define X_MIN_PIN                         24
  #endif
  #ifndef X_MAX_PIN
    #define X_MAX_PIN                         22
  #endif
#endif
#ifndef Y_STOP_PIN
  #ifndef Y_MIN_PIN
    #define Y_MIN_PIN                         28
  #endif
  #ifndef Y_MAX_PIN
    #define Y_MAX_PIN                         26
  #endif
#endif
#ifndef Z_STOP_PIN
  #ifndef Z_MIN_PIN
    #define Z_MIN_PIN                         30
  #endif
  #ifndef Z_MAX_PIN
    #define Z_MAX_PIN                         32
  #endif
#endif

----------------------------------------------------------------------------------------

The Min and Max Pins are reversed to the numbers  for Marlin 2. I changed that and it worked! 

Best regards,

Stephan

derstephan2021, 2021-01-25:
I think, I ran into a connection problem which might have caused the problem

closed 2021-01-25 by derstephan2021

Vertabreak, 2021-01-25:
Last I tested it was normal on my mecreator 2 I can retest if needed to confirm.