MarlinFirmware/Configurations #408: Need Ender 3 w/ SKR E3 mini v2 firmware w/ Runout Sensor

RetArsonFD, 2021-01-30:
Does anybody have a working firmware for an Ender 3 with a SKR E3 mini v2 that has the filament runout sensor activated ?  I've got a simple, end stop switch style sensor.  Just let me know what port you have programmed to run the sensor.  I tried to plug it into the TFT35, but that didn't work.
THANKS !!!!

sarvenn, 2021-02-08:
I use my runout sensor plugged to my tft 35. Please update your TFT firmware and open the filament runout sensor option in the Feature menu of TFT. If needed invert the logic. You are good to go. On the other hand you need some features to be activated in your marlin config for proper usage of the feature.

RetArsonFD, 2021-02-08:
> I use my runout sensor plugged to my tft 35. Please update your TFT firmware and open the filament runout sensor option in the Feature menu of TFT. If needed invert the logic. You are good to go. On [...]

Mine is plugged into my TFT35, and I thought I had updated the firmware to run it......but apparently not.  I'd rather it be plugged into the TFT, but if I can't make it work that way, I think I'll take it off all together.  It's getting to be more of a pain in the ass thn it's worth trying to make it work.  Could you please show me EXACTLY what you did to update the TFT35 firmware to make it work ?  THANKS !!!!!

sarvenn, 2021-02-09:
Get your pre-compiled firmware from this repo: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware
You need to copy your TFT specific firmware.bin file, config.ini and theme files (logos and fonts) to your SD card.
They are all under "Copy to SD Card root directory to update" folder.
In TFT 35 feature menu, you need to enable filament run out sensor option, if you use BTT smart sensor then choose smart.
There is also a logic invert option. In my case I needed to invert the logic. That's all.

Please read the README file in the main page of the repository for further info.
You also need to configure your marlin for better TFT experience.
They all documented in the README section. Also open the config.ini file in notepad. You will see those kind of instructions as well. Which features should be enabled in marlin configurations etc.

thisiskeithb, 2021-02-22:
Use the [SKR Mini E3 V2 Ender-3 example](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%202.0) and enable `FILAMENT_RUNOUT_SENSOR`.

According to the [SKR Mini E3 V2 pins diagram](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0-PIN.pdf), you'll plug your filament runout sensor into `PC15`.

closed 2021-02-22 by thisiskeithb