MarlinFirmware/Configurations #410: [BUG] (short description)

mightystormy, 2021-02-02:
### Bug Description

Autobed Leveling not working on Creality Ender 3 Pro with board 4.2.7 and BLTouch v3.1

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Configurations/files/5913660/Marlin.zip)

### Steps to Reproduce

1. Add G29 after G28
2. Print some test print

**Expected behavior:**

The print head is being adjusted in order to correct print bed irregularities.

**Actual behavior:**

There is no visible improvement when printing with G29 or printing with G29.
When G29 is added in the code, the auto-level sequence is executed by probing 9 points. Yet when the printing starts the mesh information is just ignored.

ellensp, 2021-02-09:
zip file is actually a .rar, so unless you using rar for all compressed files this wont open on most machines.

ellensp, 2021-02-09:
version is 020007, this need to be tested on bug fix first.

mightystormy, 2021-02-09:
I have updated the file with a proper zip archive.

The configuration files were teste on the latest bugfix with the same results.

mightystormy, 2021-02-09:
> version is 020007, this need to be tested on bug fix first.

The configuration files were teste on the latest bugfix with the same results.

mightystormy, 2021-02-09:
> zip file is actually a .rar, so unless you using rar for all compressed files this wont open on most machines.

I have updated the file with a proper zip archive.

ellensp, 2021-02-09:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

rechena, 2021-02-13:
The same is happening to me, I've added some more details on the other BUG open which I assume is a duplicate of this or vice versa. - https://github.com/MarlinFirmware/Marlin/issues/20922

mightystormy, 2021-02-21:
Sorry, I have seen the replies too late, I have already remove the BLTouch and disabled ABL in the firmware. The thought of restarting entire ABL saga is not appealing for the time being.

Below are the information I've added to the bug linked by rechena, maybe will help someone somehow:

I have tried both bilinear and UBL with same results, meaning no visible improvement to the test pattern. I wanted to try firmware Marlin 2.0.3 or 2.0.5 as I've read reports that the ABL still works in those versions, but I was unable to configure them for Ender 3 Pro v4.2.7 board. If somebody can try this, it is worth a shot.
Meanwhile I have just given up to bed leveling for the moment. I have decided to remove the BLTouch and do a manual bed leveling. It worked from the start and it provides consistent results. For me this means there is no mechanical issue at play concerning the printer or bed. Also the BLTouch self 10x probing test reports very small deviation which is well in the margin for ABL. In the end the fault lies within firmware somewhere.
I wanted ABL because from time to time I need to fine tune the level of the bed for small deviations and I wanted those small tunings to be handled automatically. But the time wasted trying to figure out why ABL is not working vs 4min tuning @ roughly 2 days .... yeah, I am just gonna enjoy my prints.

PS. Maybe it's useful to add that I have also tried the last TH3D Unified firmware with same result -- no ABL working.

closed 2021-02-21 by mightystormy

mightystormy, 2021-02-21:
Also, a short feedback about what was my expectation when I opened the issue.
 
I thought that, if the firmware supports my configuration, than it means is was successfully tested on my configuration. So when I explained how to reproduce the results, my expectation was that those steps will be checked on the printer that validated the configuration and I would receive a status saying our validation system does not confirm the problem, let's gather more information to understand what is happening on your end.
But if you start from scratch, based only on user data, when the issue is related to a major FW functionality... I wonder if the FW was validated on all system it claims it covers. I know it is a free service and I appreciate it, but if it was clarified that ABL does not  work for Ender3-Pro with board v4.2.7, I would have not bought the BLTouch in the first place.