MarlinFirmware/Configurations #411: [FR] Missing platformio environment

X-Ryl669, 2021-02-04:
### Bug Description

Would it be possible to include a file called `platformio.env` in each configuration folder, containing the expected platformio environment to pass to build the repository (the same as what to use in `platformio run --environment <here>`. This would be really helpful to automate testing that all configuration builds with some Marlin's PR. Currently, one has to guess what environment to use for one configuration.

I've tried to use `Marlin/src/core/boards.h` to match the board with something else but it's not clear at all.
 

#### Additional Information

I'm preparing a PR for building Marlin with all configuration in this repository. I'll update this post when it's submitted.

thisiskeithb, 2021-02-22:
Is this still needed?

I believe this was answered in the main Marlin repo, but you can build with `mftest -b` to build the auto-detected environment.

Environment hints can be found in [pins.h](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/pins.h). Ex:

```cpp
#elif MB(BTT_SKR_V1_1)
  #include "lpc1768/pins_BTT_SKR_V1_1.h"        // LPC1768                                env:LPC1768
#elif MB(BTT_SKR_V1_3)
  #include "lpc1768/pins_BTT_SKR_V1_3.h"        // LPC1768                                env:LPC1768
#elif MB(BTT_SKR_V1_4)
  #include "lpc1768/pins_BTT_SKR_V1_4.h"        // LPC1768                                env:LPC1768
```

X-Ryl669, 2021-02-22:
Nope, you're right, it's done now in Marlin's repo thanks to the new `build_all_examples` script.

closed 2021-02-22 by X-Ryl669