MarlinFirmware/Configurations #413: [BUG] CrealityV422 config from bugfix-2.0.x not compilable with Marlin bugfix-2.0.x

maxgerhardt, 2021-02-06:
### Bug Description

When using the configuration from 

https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3/CrealityV422

with the Marlin branch 

https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

Many compile errors arise.

### Configuration Files

As per https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3/CrealityV422

### Steps to Reproduce

1. Download Marlin in the configuration files from the above links
2. Put the 4 header files (`Configuration.h`, `Configuration_adv.h`, `_Bootscreen.h`, `_Statusscreen` in the Mariln subfolder, overwriting the old files
3. Change the `platformio.ini` to do `default_envs = STM32F103RET6_creality`
4. Press Build

**Expected behavior:**

Firmware compiles since both Marlin and the used config files come from the same branch name.

**Actual behavior:**

Build failure. First error is 

```
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
In file included from buildroot/share/PlatformIO/scripts/common-dependencies.h:48:0:
c:\users\max\documents\marlin\marlin\configuration_adv.h:1069:19: error: #if with no expression
   #if IS_ULTIPANEL
                   ^
CalledProcessError: Command 'arm-none-eabi-g++.exe -DTARGET_STM32F1 -D__MARLIN_FIRMWARE__ -DSTM32_XL_DENSITY -DTEMP_TIMER_CHAN=4 -D__MARLIN_PREBUILD__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependenc
```

which, as per https://github.com/MarlinFirmware/Configurations/issues/272#issuecomment-710814493, can be fixed by replacing a block of code in `Configuration_adv.h`. 

Then you run into the error 

```
In file included from Marlin\src\module\../inc/MarlinConfig.h:39:0,
                 from Marlin\src\module\motion.h:31,
                 from Marlin\src\module\motion.cpp:27:
Marlin\src\module\motion.cpp:151:16: error: 'HOMING_FEEDRATE_XY' was not declared in this scope
     MMM_TO_MMS(HOMING_FEEDRATE_XY), MMM_TO_MMS(HOMING_FEEDRATE_XY),
                ^
```

which, refering to the issue https://github.com/MarlinFirmware/Configurations/issues/378 means that the configuration and Marlin versions / branches were different. But, they're not, they're downloaded from the same branch name.

So, for the CrealityV422, the pushed configuration files in bugfix-2.0.x are not compatible with Marlin from the same branch, in my eyes. 

#### Additional Information
-/-

thisiskeithb, 2021-02-06:
I just took the latest Marlin [`bugfix-2.0.x` (ee66d9c)](https://github.com/MarlinFirmware/Marlin/archive/ee66d9ccf9772206d151cf821b5adff2081423bb.zip) (**not `2.0.x`**) and dropped the referenced Ender-3 with 4.2.2 board configs from [`/examples/Creality/Ender-3/CrealityV422`](https://github.com/MarlinFirmware/Configurations/tree/e29ea3829e1d98b3c636052411c719c00da18e2e/config/examples/Creality/Ender-3/CrealityV422) and they build fine:
```prolog
Building .pio/build/STM32F103RET6_creality/firmware-20210206-152509.bin
Checking size .pio/build/STM32F103RET6_creality/firmware-20210206-152509.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  16.5% (used 10808 bytes from 65536 bytes)
Flash: [===       ]  31.4% (used 164836 bytes from 524288 bytes)
============================== [SUCCESS] Took 21.20 seconds ==============================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  SUCCESS   00:00:21.202
=============================== 1 succeeded in 00:00:21.202 ===============================

Terminal will be reused by tasks, press any key to close it.
```

Your config & Marlin versions needs to match. `bugfix-2.0.x`/`import-2.0.x` Configuration branches are only compatible with Marlin `bugfix-2.0.x` and **not `2.0.x`**.

closed 2021-02-06 by thisiskeithb

maxgerhardt, 2021-02-06:
Well I was using `bugfix-2.0.x` but upon closer inspection a not-updated version of the branch. Was somewhere in september last year or something. The `git pull` didn't work as I wanted to :/ 

I'm redownloading Marlin and switching to the `bugfix-2.0.x` branch to restest it on my side.