MarlinFirmware/Configurations #414: BTT SKR 1.4 TURBO SU ENDER 3 IMPOSSIBILE COMPILARE MARLIN

Gioele2803, 2021-02-07:
Ciao a tutti è ormai da giorni che provo a configurare Marlin v2.0.7.2  con mainboard SKR 1.4 turbo
ho problemi sempre con le stesse cose
i motori sono TMC2208
questo è quello che mi spunta

PS C:\Marlin-2.0.7.2> platformio run -e LPC1769
Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1769; framework: arduino)
-----------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.6
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <TMCStepper> 0.7.1
|-- <U8glib-HAL> 0.4.3
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1769/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1769\src\src\module\motion.cpp.o
Compiling .pio\build\LPC1769\src\src\module\stepper.cpp.o
Compiling .pio\build\LPC1769\src\src\module\stepper\trinamic.cpp.o
Compiling .pio\build\LPC1769\src\src\module\temperature.cpp.o
In file included from Marlin\src\module\../inc/MarlinConfig.h:47,
                 from Marlin\src\module\motion.h:31,
                 from Marlin\src\module\motion.cpp:27:
Marlin\src\module\motion.cpp:151:16: error: 'HOMING_FEEDRATE_XY' was not declared in this scope; did 
you mean 'HOMING_FEEDRATE_MM_M'?
  151 |     MMM_TO_MMS(HOMING_FEEDRATE_XY), MMM_TO_MMS(HOMING_FEEDRATE_XY),
      |                ^~~~~~~~~~~~~~~~~~
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
   76 | #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
      |                                           ^~~~
Marlin\src\module\motion.cpp:151:48: error: 'HOMING_FEEDRATE_XY' was not declared in this scope; did 
you mean 'HOMING_FEEDRATE_MM_M'?
  151 |     MMM_TO_MMS(HOMING_FEEDRATE_XY), MMM_TO_MMS(HOMING_FEEDRATE_XY),
      |                                                ^~~~~~~~~~~~~~~~~~
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
   76 | #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
      |                                           ^~~~
Marlin\src\module\motion.cpp:153:14: error: 'HOMING_FEEDRATE_Z' was not declared in this scope; did you mean 'HOMING_FEEDRATE_MM_M'?
  153 |   MMM_TO_MMS(HOMING_FEEDRATE_Z)
      |              ^~~~~~~~~~~~~~~~~
Marlin\src\module\../inc/../core/types.h:76:43: note: in definition of macro 'MMM_TO_MMS'
   76 | #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)
      |                                           ^~~~
Marlin\src\module\stepper\trinamic.cpp:214:27: error: 'X_SLAVE_ADDRESS' was not declared in this scope
  214 |       TMC_UART_DEFINE(SW, X, X);
      |                           ^
Marlin\src\module\stepper\trinamic.cpp:57:145: note: in definition of macro 'TMC_UART_SW_DEFINE'
   57 | #define TMC_UART_SW_DEFINE(IC, ST, L, AI) TMCMarlin<IC##Stepper, L, AI> stepper##ST(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN, float(ST##_RSENSE), ST##_SLAVE_ADDRESS)
      |
                                                   ^~
Marlin\src\module\stepper\trinamic.cpp:63:39: note: in expansion of macro '_TMC_UART_DEFINE'
   63 | #define TMC_UART_DEFINE(SWHW, ST, AI) _TMC_UART_DEFINE(SWHW, ST##_DRIVER_TYPE, ST, AI##_AXIS)      |                                       ^~~~~~~~~~~~~~~~
Marlin\src\module\stepper\trinamic.cpp:214:7: note: in expansion of macro 'TMC_UART_DEFINE'
  214 |       TMC_UART_DEFINE(SW, X, X);
      |       ^~~~~~~~~~~~~~~
Marlin\src\module\stepper\trinamic.cpp:232:27: error: 'Y_SLAVE_ADDRESS' was not declared in this scope
  232 |       TMC_UART_DEFINE(SW, Y, Y);
[... 50 lines cut ...]
c:\marlin-2.0.7.2\marlin\src\core\macros.h:194:29: note: in expansion of macro '__TERN'
  194 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'    
      |                             ^~~~~~
c:\marlin-2.0.7.2\marlin\src\core\macros.h:191:29: note: in expansion of macro '_TERN'
  191 | #define TERN0(O,A)          _TERN(_ENA_1(O),0,A)    // OPTION converted to A or '0'
      |                             ^~~~~
Marlin\src\module\stepper\trinamic.cpp:851:129: note: in expansion of macro 'TERN0'
  851 |   #define TMC_SW_DETAIL_ARGS(A) TERN(A##_HAS_SW_SERIAL, A##_SERIAL_TX_PIN, -1), TERN(A##_HAS_SW_SERIAL, A##_SERIAL_RX_PIN, -1), TERN0(A##_HAS_SW_SERIAL, A##_SLAVE_ADDRESS)
      |
                                   ^~~~~
Marlin\src\module\stepper\trinamic.cpp:852:28: note: in expansion of macro 'TMC_SW_DETAIL_ARGS'      
  852 |   #define TMC_SW_DETAIL(A) TMC_SW_DETAIL_ARGS(A)
      |                            ^~~~~~~~~~~~~~~~~~
Marlin\src\module\stepper\trinamic.cpp:854:5: note: in expansion of macro 'TMC_SW_DETAIL'
  854 |     TMC_SW_DETAIL(X), TMC_SW_DETAIL(X2),
      |     ^~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\LPC1769\src\src\module\stepper\trinamic.cpp.o] Error 1
==================================== [FAILED] Took 9.89 seconds ==================================== 

Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:09.886
=============================== 1 failed, 0 succeeded in 00:00:09.886 ===============================PS C:\Marlin-2.0.7.2> echo "done" >C:\Users\UTENTE~1\AppData\Local\Temp\ipc
PS C:\Marlin-2.0.7.2>

ellensp, 2021-02-07:
With Marlin-2.0.7.2 you need to use Configuration files for 2.0.7.2 
The config file you are using is either the  bugfix or for import version should be release.
Use these configs https://github.com/MarlinFirmware/Configurations/archive/release-2.0.7.2.zip

closed 2021-02-07 by ellensp

ellensp, 2021-02-07:
For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

Gioele2803, 2021-02-07:
Non cambia nulla, non va lo stesso