MarlinFirmware/Configurations #415: [BUG] Bootloop after install

djonas171, 2021-02-10:
### Bug Description
After a compiled version of Marlin `bugfix-2.0.x` has been installed on a M200 v2 (STM32F070CB) the printer ends up in a bootloop.
Here everything works (motors endstop etc. ) for approx. 5 seconds then the system restarts and then it boots up and works for 5 seconds again.
 

### Configuration Files
The files used are the files in the marlin example: https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Malyan/M200
I thought it was a problem with the watchdog, so i have also tried without watchdog enabled under _adv, but with no luck

### Steps to Reproduce

1. 
Download newest bugfix release, change the config files. (i have tried with marlin 2.0.7 with the same results)
2. 
Compile without errors and upload the .bin file to the board
3.
turn on the power. 

**Expected behavior:**

The printer turns on and stays on. 

**Actual behavior:**

The printer turns on, then shutdown after 5 sec. Then turns on again. 

#### Additional Information

When the printer is on it works, but only for a 5 second period.
When i flash the printer with the bin file that came with it, it works without any problems

Hope someone can help me - Jonas

ellensp, 2021-02-17:
Never ever use import branch config files
Use bugfix branch configs for bugfix code or release branch configs for release versions.

Try again with latest bugfix and bugfix config examples

thinkyhead, 2021-04-25:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- Marlin Discord server. Join link: https://discord.gg/n5NJ59y
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Tom's 3D Forums](https://forum.toms3d.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-04-25 by thinkyhead