MarlinFirmware/Configurations #421: [FR] Configuration for Ender 3 MAX

spooky79, 2021-02-21:
hi !
Will there be any configuration for Ender 3 MAX with silent board 4.2.2  (TMC2208 chip driver is adopted to use in X-axis and Y-axis) or skr E3 mini v1.2 (or v2.0) ?

devildj3400, 2021-02-24:
Seguo ! Anche io cerco le configurazioni per ender 3 Max ! Grazie mille

spooky79, 2021-03-08:
Po wielu godzinach testów  skompilowałem wersję działającą, przetestowaną  . Jest to wersja od "ender 3 pro " z płytą główną V4.2.2 . Do zmiany :  ustawić  poprawną płytę główną  (creality V4.2.2) ,    pole robocze stołu  , sterowniki  X,Y TMC2208_standalone , sterowniki Z,E na A4988 ,  PID BED oraz PID HOTEND , uaktywniamy FILMANET RUNOUT .  Gotowe :)

spooky79, 2021-03-08:
a working version  compiled on the basis of "Ender 3 Pro V 4.2.2"

closed 2021-03-08 by spooky79