MarlinFirmware/Configurations #422: [BUG] [TwoTrees Sapphire Plus] - ENDSTOP_INTERRUPTS_FEATURE

dexter323i, 2021-02-21:
Printer is TwoTrees Sapphire Plus, MKS Robin Nano V1.2 mainboard, the Firmware is Marlin 2.0.7.2.
I used config based on import-2.0.x branch. The homing was not reliable. Usually the Y limit switch clicked, but the motors still wanted to go further. Sometimes worked okay, sometimes the quick homing phase was okay, but the second slower homing did not trigger, and FW went into fault state that homing is not possible. And sometimes the quick moving homing phase was wrong, and the motor wanted to go forever. I checked the endstop states with M119, and it showed the states perfectly. Always perfectly. Not an issue there, but homing was not reliable.

Then I turned off the ENDSTOP_INTERRUPTS_FEATURE, and it repaired. Now the homing works flawlessly. I suggest not to enable this feature on TwoTrees Sapphire Plus printers and/or MKS Robin Nano V1.2 mainboards.

[MarlinConfigs.zip](https://github.com/MarlinFirmware/Configurations/files/6017698/MarlinConfigs.zip)
(This is the bad config)

Video:
https://drive.google.com/file/d/1dBsQRLu4z1M1Sk8sDBZkrTZkvRd7aaLR/view?usp=sharing

ellensp, 2021-02-23:
These board have noisy endstops, you should enable ENDSTOP_NOISE_THRESHOLD

thisiskeithb, 2022-08-10:
@dexter323i Did you re-enable `ENDSTOP_INTERRUPTS_FEATURE ` & enable `ENDSTOP_NOISE_THRESHOLD`?

dexter323i, 2022-08-10:
@thisiskeithb I don't know. Switched to MTT SKR-2 board ages ago. Robin nano is history. Should I close this issue?

thisiskeithb, 2022-08-10:
No worries. I'll close it.

closed 2022-08-10 by thisiskeithb