MarlinFirmware/Configurations #426: Crealityv422 - error in line [BUG] (short description)

RudyBenner, 2021-02-24:
### Bug Description

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

**Actual behavior:**

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

RudyBenner, 2021-02-24:
source 

https://github.com/MarlinFirmware/Configurations/blob/b54e18d52878ed7e86820a08925edb24c7fbdc5d/config/examples/Creality/Ender-3 Pro/CrealityV422/Configuration.h

line 682 - commenting this line out stops all the warnings when compiling. Marlin 2.0.* and in Bugfix-2.0.*

ellensp, 2021-02-24:
`#warning "Creality 4.2.2 boards may have A4988 or TMC2208_STANDALONE drivers. Check your board and make sure to select the correct DRIVER_TYPE!"`

This line is meant to be there.
So you don't blindly compile without finding out if your board has the A4988, or the TMC2208_STANDALONE stepper drivers.

closed 2021-02-24 by ellensp